WEC-Sim/WEC-Sim #201: Unclear on the process

rubintaipi1, 2017-09-02:
Hello, 

We are a bit unclear on how the process goes from the moment we finish up the model in solidworks to the moment we are simulating in wec-sim. 

We are using NEMOH for our BEM calculations and that takes in a .dat file but solidworks puts out a .stl file.  How do we go about this? Is there a way to get solidworks to give us .dat files of our model so we can input them into NEMOH?

We can't use wammit because the demo version on their site wont run on my windows 10 machine. 

If I am not mistaken, after NEMOH does the calculations, we feed those files into BEMIO and that will create the HDF5 files. We then take those HDF5 files and refer to them in our wec-sim input file. 

Any help would be appreciated.

dforbush2, 2017-09-05:
Hello, 

An stl mesh can be converted to one usable by NEMOH using the python version of bemio.
For installation instructions see https://wec-sim.github.io/bemio/installing.html

For specific instructions on mesh conversion, please consult the tutorial section https://wec-sim.github.io/bemio/module_doc.html#bemio-mesh-utilities

This functionality is in progress, but does not yet exist in the matlab version of BEMIO hosted on the WEC-sim repository.

-Dom

kmruehl, 2017-09-05:
@rubintaipi1, thanks for posting this question. We recently added a [workflow diagram](http://wec-sim.github.io/WEC-Sim/overview.html) to the wec-sim website to help address process questions, but I agree the WEC-Sim pre-processing steps could use more detail. 

In short, @dforbush2's answer is correct. The mesh conversion from a SolidWork's generated `*.stl` file to a NEMOH mesh is currently included in the python version of BEMIO - and we are in the process of adding it to the version of BEMIO included in the WEC-Sim release. Your question directly relates to this feature request, a.k.a. 'project': https://github.com/WEC-Sim/WEC-Sim/projects/10 

Also worth nothing is that the `*.stl` meshes generated from SolidWorks are generally _very_ coarse, and I recommend checking its resolution before using it to run NEMOH. A coarse mesh will impact the quality of your BEM solution. For more information about generation of `*.stl` meshes, check out our [website here](http://wec-sim.github.io/WEC-Sim/advanced_features.html#stl-file-generation), and the [non-linear hydrodynamics webinar](http://wec-sim.github.io/WEC-Sim/webinars.html#webinar-2-nonlinear-hydro-non-hydro-and-b2b) (non-linear hydro uses the `*.stl` mesh for force calculations).

kmruehl, 2017-09-11:
@rubintaipi1 does this answer your question?

rubintaipi1, 2017-09-19:
Hi Kelly,

Sorry I have not gotten back to you sooner. The answer you provided was
very helpful!
I was able to generate the .dat files from the methods provided in the
python version of BEMIO.

I am now fighting with NEMOH to generate the .h5 files for input into
wecsim.

Thanks again!

rubintaipi1, 2017-09-19:
Kelley*

sorry for the typo :)

rubintaipi1, 2017-09-19:
Hello again,

I am having some problems getting the NEMOH simulation to run and as I am
investigating, I am seeing that in the example files (the cylinder) the dat
file in the mesh folder has entries for each node and each panel. The .dat
file that BEMIO generated from my .stl conversion only has one set of
entries. Could this be why I am not able to run the NEMOH simulation?

Would I be able to get on a 15 min call with you just to explain and maybe
show you what I mean? If so, it would be a great help.

Thanks so much!
-Rubin

dforbush2, 2017-09-19:
Hi Rubin, 
I have a couple of questions/suggestions for you. From your description, it sounds like BEMIO is outputting a file formatted like a WAMIT gdf (which only has the nodes and not the connectivity map) but it is somehow saving as a NEMOH .dat. 

In your script to execute BEMIO you should have a line that looks like:

# read file
YOUR_VARIABLE=mesh.read(file_name='your_file.stl') 
#specify output format
YOUR_VARIABLE.write(mesh_format='NEMOH')

What are the first few lines of the output file bemio gave you? In the list of nodes, (the first array), is it 3 or 4 columns wide? The nemoh format for this array should be four columns wide 

[node_num     node_numA     node_numB     node_numC]

so I am curious to see if this formatting is correct.

(A word of caution: STL files use a triangular panel, while NEMOH and WAMIT use a quadrangular panel. Converting .stl to any other mesh format makes a degenerate quadrangle from each triangle (i.e., the first and last points are the same, ABC --> ABCA). This should not be what is preventing NEMOH from running, however.),

rubintaipi1, 2017-09-19:
Hi,

Yes that is correct. That is the script that I ran. Here is a few of the
top lines and a few of the bottom lines.

As you can see by the number of lines in the file and the number on the
row, the count does not start again as it does in the cyinder.dat example
file, also attached here.

What i noticed in the nemoh.cal file is that the number of nodes and panels
refers to the numbers in the .dat file. For example, in the cylinder.dat
file they are 2016 and 504.
Note line 2017 on cylinder.dat.

I hope that makes sense.

Thanks!

dforbush2, 2017-09-19:
Rubin,

I do not actually see any attached files. I am not sure attachments sent via email will end up on this thread. Please submit any attachments directly to this thread.

Thank you,
Dom

rubintaipi1, 2017-09-19:
I updated my previous comment. Had to rename the cylinder.dat to .txt since dat is not allowed on github..

kmruehl, 2017-09-19:
@rubintaipi1 you can upload *.zip files here

rubintaipi1, 2017-09-20:
Ok so here I am attaching a zip of the files i am trying to run NEMOH with and the cylinder.dat file that I am comparing my dat file to. My .dat file is in the mesh folder. 
[upperModule.zip](https://github.com/WEC-Sim/WEC-Sim/files/1318381/upperModule.zip)

dforbush2, 2017-09-20:
Hi Rubin,

I took a look at your mesh and, as it is not properly formatted, I am not surprised that NEMOH could not run it. Without having your input .stl file, it is hard for me to deduce what the bemio side issue is, although it seems like the run is terminating prematurely before it finishes writing the file. Namely, see line 993 of mesh.py: this seems to be as far as you're getting and the next for loop is not executing for some reason. If the faces of your .stl file are not properly identified within 'read_stl' (i.e., the xrange(self.faces.shape[0]) is zero) this could happen, which may indicate that there is a problem in the .stl file format. Is it possible to share this file?

We are in the process of moving this bemio functionality to Matlab and it is very nearly there, but I am still chasing a couple of bugs. 

Thanks,
Dom

rubintaipi1, 2017-09-20:
Hi Dom, 

Is there a place where i can send you these files privately? I don't want to expose these to the whole internet. :)

dforbush2, 2017-09-20:
Hi Rubin,

Instead, I have uploaded meshConvert.m to the BEMIO subfolder of the WECSIM distribution. I believe it has all the functionality you will need to convert your STL to a DAT mesh. If you prefer to simply download this single file, make sure to put it in the /WECSIM/source/functions/BEMIO directory.

From the command line;
On windows:
OutputVarName=meshconvert('your_input.stl','stl','dat','outputfilename',2,0);

On mac/Linux
OutputVarName=meshconvert('your_input.stl','stl','dat','outputfilename',1,0);

(The distinction by operating system has to do with the new line delimiters in an stl file)

I've tested this with a handful of stl files I have and it works great, so hopefully it does for you as well! Please let me know if you encounter problems with meshConvert.m.

-Dom

rubintaipi1, 2017-09-21:
Hi Dom, 

Please see what I am getting while trying to run the new converter. 

[image]

I think it would probably make sense if you saw the stl files that we are trying to work with. Please let me know if you agree and how we would go about that. 

Thanks!
Rubin

dforbush2, 2017-09-21:
Rubin,

I think I may have figured it out. It appears you are attempting to run a binary-formatted STL file? At the current time both the python and matlab versions of Bemio only support ASCII formatted STL files. It should be a simple matter to re-output your stl model in an ASCII format. This is stated in the comments of the meshConvert code, but I will ensure this is made clear in the eventual accompanying documentation.

This is a little speculative, because I have not seen your model, but I get an identical error if I try to run meshConvert.m from a binary-formatted STL, so it has some promise! If for some reason converting to ASCII is not an option, I would take a look at 

https://www.mathworks.com/matlabcentral/fileexchange/22409-stl-file-reader 

to handle your binary file (please note that I cannot offer support on this code). It has a similarly formatted output to the one currently used by meshConvert.m, so it should only require minimal tweaking to make it work.

If this does not fix your issue let me know and we can work out a way to transfer your STL directly.

Best of Luck,
Dom

rubintaipi1, 2017-09-21:
Hi Dom, 

I went ahead and regenerated the stl files in ascii format and that seemed to help get me further but for now I am getting an error with fopen. I tried to do a bit of digging as to what the problem could be but got nothing. I added a check to see if fopen was returning -1 i.e. unable to open the file. 

Please see the screenshot below
[image]

rubintaipi1, 2017-09-21:
Thanks for all your help on this by the way! It is greatly appreciated!!

dforbush2, 2017-09-21:
Hi Rubin,

As per the example, the 'outputfilename'  should not include the file-type extension.

In your example, your fourth argument input should be 'upperModuleSmaller', NOT 'upperModuleSmaller.dat'. The extension is added on line 210.

Dom

rubintaipi1, 2017-09-21:
Hi Dom, 

I just tried it again without the extension at the end and still got the same error. I also removed the check that I added for consistency and too a screenshot of what I am getting. 

[image]

Thanks!
Rubin

kmruehl, 2017-09-21:
@rubintaipi1 I don't believe the error you are experiencing is with the BEMIO code, since we are unable to reproduce the error on our end, I don't believe we can be of any further help. 

@dforbush2 thanks for your help in trying to resolve this.

closed 2017-09-21 by kmruehl

rubintaipi1, 2017-09-22:
Thank you both for your support on this. Once I figure it out, I will post what the issue was.