WEC-Sim/WEC-Sim #203: B2B: Getting errors when running irregular waves

ratanakso, 2017-09-11:
Hello,

I was trying to run a simple test case with RM3 only moving in the heave direction (i.e., translational constraint is used). I turned on the body-to-body interactions (simu.b2b=1) and I ran irregular waves (i.e., waves.H=2.5, waves.T = 8). PTO was set to zero. I got the following errors:
Error using wecSim (line 240)
Error due to multiple causes.

Caused by:
    Error using wecSim (line 240)
    Simulink cannot unify the dimensions of the multiplexed signal at
    output port 1 of 'RM3/Float/Hydrodynamic Body/Wave Radiation
    Forces Calculation/SS CI and Constant-Damping-CoeVariant
    Subsystem/Convolution Integral Calculation/MATLAB Function/v' to
    match the input dimension of input port 1 of
    'RM3/Float/Hydrodynamic Body/Wave Radiation Forces Calculation/SS
    CI and Constant-Damping-CoeVariant Subsystem/Convolution Integral
    Calculation/MATLAB Function/ SFunction '. Specifically, Simulink
    cannot unify {6,6} to match [6x1]. Consequently, Simulink cannot
    propagate dimension information from output port 1 of
    'RM3/Float/Hydrodynamic Body/Wave Radiation Forces Calculation/SS
    CI and Constant-Damping-CoeVariant Subsystem/Convolution Integral
    Calculation/MATLAB Function/v' to input port 1 of
    'RM3/Float/Hydrodynamic Body/Wave Radiation Forces Calculation/SS
    CI and Constant-Damping-CoeVariant Subsystem/Convolution Integral
    Calculation/MATLAB Function/ SFunction '.
    Error using wecSim (line 240)
    Error in port widths or dimensions. Invalid dimension has been
    specified for input 'v'.
    Error using wecSim (line 240)
    Error occurred in 'RM3/Float/Hydrodynamic Body/Wave Radiation
    Forces Calculation/SS CI and Constant-Damping-CoeVariant
    Subsystem/Convolution Integral Calculation/MATLAB Function'.

If I switch to regular waves, the simulation works! It seems like it is an issue with irregular waves when the body-to-body interactions option is used. Could you please check?

Thanks!

kmruehl, 2017-09-14:
@ratanakso There should not be an issue running WEC-Sim with irregular waves. Can you zip the model and upload it here? If it's the RM3, you may only need to upload the `wecSimInputFile.m`. Thanks!

ratanakso, 2017-09-18:
@kmruehl I have attached the slx and m file. Thanks!
[RM3wB2B.zip](https://github.com/WEC-Sim/WEC-Sim/files/1311604/RM3wB2B.zip)

kmruehl, 2017-09-18:
@ratanakso Thanks for sharing this. It looks like we never modified the radiation CIC calculation in `ConvolutionIntegral_interp.m’ to` account for B2B interaction. B2B currently works for constant radiation damping (regular waves, etc), and for state space (`simu.ssCalc = 1`). We will work on revising it for the radiation CIC calculation ASAP.

ratanakso, 2017-09-18:
@kmruehl Thank you for looking into this issue.

closed 2017-10-12 by kmruehl