WEC-Sim/WEC-Sim #204: Modelling of a floating buoy constrained to heave movement!

dshort, 2017-09-18:
Firstly, to the developers: Thank you so much for producing such a brilliant piece of open source software for the wave energy community. 
I am attempting to simply model a heaving cylindrical buoy without a power take off to validate investigate wave induced forces and heave response. I have successfully (I think!) performed the frequency domain calcs for hydrodynamic coeffs using Nemoh and have set up the simulink model with one hydrodynamic body connected to the global reference frame through the translational constarint. 
The wecSim command runs through and i get plots for  heave response and surface elevation however the following errors are seen:

Index exceeds matrix dimensions.

Error in responseClass/plotResponse (line 128)
            t=obj.bodies(bodyNum).time;

Error in userDefinedFunctions (line 11)
output.plotResponse(2,3);

Error in wecSim (line 251)
    userDefinedFunctions;

Also the visualisation Mechanics looks very strange. The water surface is very small and the body is enormous!! If you have any suggestions regarding where I am goibng wrong I would be hugely grateful.

kmruehl, 2017-09-18:
Thank you for your positive feedback about wec-sim, it is much appreciated.

In the `userDefinedFunctions` script, when you call on `output.plotResponse(2,3);`, you are plotting the heave response of body(2), however your model only has body(1) defined for the heaving cylinder. The error you received is from the `plotResponse` function in `responseClass.m` line 128. If you change script to  `output.plotResponse(1,3);`, to plot the heave response of body(1), you should no longer get this error. 

see  `responseClass.m` lines 123-128:
```
        function plotResponse(obj,bodyNum,comp)
            % Plots response of a body in a given DOF
            %   'bodyNum' is the body number to plot
            %   'comp' is the response direction to be plotted (1-6)
            DOF = {'Surge','Sway','Heave','Roll','Pitch','Yaw'};
            t=obj.bodies(bodyNum).time;
```

dshort, 2017-09-19:
Thank you very much Kelley that has got rid of the error. However
visualisation in matlab mechanics explorer still looks very strange. The
body is enormous compared to the water surface. Here is a screenshot.
[image: Inline image 1]

Any ideas how to get a nicer visualisation where the buoy is seen to be
floating on a large waterplane in deep water?

kmruehl, 2017-09-19:
To change the domain size, you can define the variable `simu.domainSize` in the `simulationClass.m` (line 30). The default is `simu.domainSize = 200m`.

closed 2017-09-19 by kmruehl

santi520, 2022-06-14:
> Thank you very much Kelley that has got rid of the error. However visualisation in matlab mechanics explorer still looks very strange. The body is enormous compared to the water surface. Here is a [...]

When you export the .stl file, specify that you want it in "m". I had the same issue because I design everything in "mm". Exporting as .stl in solidWorks specifying "m" solved the issue.