WEC-Sim/WEC-Sim #205: Single Body Point absorber (buoy)

haszan21, 2017-09-18:
I keep getting this error, and I don't know how to fix it, any help!?

Error using wecSim (line 236)
['Buoy/Global Reference Frame/Solver Configuration']: Error evaluating equations at time 0.0. Stopping simulation.
There may be a singularity in the solution. If not, try reducing the step size (either by reducing the fixed step size or by tightening the error tolerances)
Caused by:
Error using wecSim (line 236)
['Buoy/Floating (3DOF)/CONSTRAINT (Planar Joint)']: 'Buoy/Floating (3DOF)/CONSTRAINT (Planar Joint)' has a degenerate mass distribution on its follower side.

I have tried to change the time step but still get the error

kmruehl, 2017-09-18:
@haszan21 can you zip your model and upload it here?

haszan21, 2017-09-19:
@kmruehl thanks for your help!

[Buoy.zip](https://github.com/WEC-Sim/WEC-Sim/files/1313151/Buoy.zip)

kmruehl, 2017-09-19:
@haszan21, attached is a working version of your buoy model. I changed the following things:
- You had a `translational constraint` connected to a `translational pto` without a mass in between. This was causing the error: **['Buoy/Floating (3DOF)/CONSTRAINT (Planar Joint)']: 'Buoy/Floating (3DOF)/CONSTRAINT (Planar Joint)' has a degenerate mass distribution on its follower side.**. 
I resolved this by removing the `translational constraint` (since it was redundant).

- The time-step needed to be reduced. We recommend setting `simu.dt` <= T/100 for all WEC-Sim simulations, and when coupling WEC-Sim with PTO-Sim, `simu.dt`  needs to be a _lot_ smaller. 

- The definition of `simu.simMechanicsFile` is case sensitive, so matlab was flagging a warning. I changed `simu.simMechanicsFile = 'buoy.slx';` to `simu.simMechanicsFile = 'Buoy2.slx';` to resolve the warning and to call on the revised`*.slx` file. 

[Buoy2.zip](https://github.com/WEC-Sim/WEC-Sim/files/1314855/Buoy2.zip)

closed 2017-09-19 by kmruehl

haszan21, 2017-09-19:
@kmruehl I have downloaded it, i am gonna run it now.

thanks so much for your help!

cadu55br, 2019-07-21:
Dear Ms. Kelley Ruehl,

I have copied the revised Buoy2.zip file attached by you above.
Then I updated your Simulink2017 model "Buoy2.slx", converting it to Simulink2019 ("Buoy2.slx" and "Buoy2.slxc").

I replaced your hydro and geometry files ("ANALYSIS.h5" and "buoy.stl") by the files generated by Matlab routine axiMesh_Nemoh_Bemio for my R5m x H6m cylinder ("VCF_190625_1800_R5H6.h5" and "buoy2.dat").

My comments:
1) I was able to run wecSim but, as expected, in spite of opening a Mechanics Explorer window in Matlab, the animation was not generated (I presume that is because I use mesh file *.dat instead of *.stl). See warning in Buoy2_simulationLog.txt.

2) The main issue is that the graphics of my model generated by wecSim were identical to the ones generated for your original Buoy2, although the masses of the buoys being very different (in my case is 231763.00 kg and your original Buoy2 is 6200.28 kg). In spite of I have removed the original case directory from Matlab path, keeping just my case directory.

Could you please verify what is the problem?
Thanks in advance!
Best regards,
Carlos

PS: I have attached the file VCF_190625_1800_R5H6.zip with my results.

[VCF_190625_1800_R5H6.zip](https://github.com/WEC-Sim/WEC-Sim/files/3414831/VCF_190625_1800_R5H6.zip)

sricharanvvs, 2019-10-16:
Hi ,
I have gone through the related posts but unable to resolve. Mine is a floating WEC, have used MoorDyn. Untiill I attach or include Moordyn the simulation is running well. 
Also, Time step is 0.01, ramp time also kept small. (2019b)
-- One thing I dont understand is the locations of PTO, Mooring etc. With which reference the location is to  be entered?
-- lines.txt is also included in mooring folder: Simulation says, NaN value is generated.
-- The below error pops up irrespective of values I change in lines.txt
Error using wecSim (line 261)
Derivative of state 'FWEC3_CBHEAVE.Translational.CONSTRAINT_Prismatic_Joint.Pz.v' in block
'FWEC3_CBHEAVE/Translational/CONSTRAINT (Prismatic Joint)' at time 0.01 is not finite. The simulation will be stopped. There may
be a singularity in the solution.  If not, try reducing the step size (either by reducing the fixed step size or by tightening
the error tolerances)
I am attaching my files. Please review and help me through. 
Thanks, 
Regards,
Charan - IITM
[image]