WEC-Sim/WEC-Sim #206: paraview

stevenshine, 2017-10-01:
hi
i am  try wave simulink about wave motion in paraview
where  there is file  and i can get them about practice?
i download the WEC-Sim_Applications-master file but there is not pvd file 
thank u

kmruehl, 2017-10-02:
@stevenshine  there are two ParaView visualization [applications posted online here](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Viz). The output files generated by WEC-Sim to run a ParaView visualization are very large and thus are not included in the online applications. You'll need to run WEC-Sim for the application case of interest, and a `/vtk` directory will be created that includes the `*.pvd` file. You can then open the `*.pvd` file to create your own visualization, or open the saved ParaView `*.pvsm` state file. 

For more information about WEC-Sim visualizations and ParaView, take a look at the [visualization webinar here](http://wec-sim.github.io/WEC-Sim/webinars.html#webinar-4-mooring-and-visualization) and the [visualization documentation here](url).

stevenshine, 2017-10-03:
thanks you
@kmruehl  
i still have some questions
1.when i run wecsim-master but it get error by matlab 
the wecs-master that i downlad from ur link( applications posted online here).
2.the two visualization files  input the paraview sofwave , there are error messages 
how to run them ?
thanks you again
[image]
[image]

kmruehl, 2017-10-03:
The Matlab error you received is because the `*.h5` files were not downloaded properly. They are stored by GitHub using Git Large File Storage (git-lfs), so you'll need to follow the [git-lfs instructions in the note here](http://wec-sim.github.io/WEC-Sim/getting_started.html#option-3-static-code-download-not-recommended).

As for the ParaView error, it's likely because the WEC-Sim run was not properly completed. Also, make sure you have loaded the WEC-Sim macros in ParaView and if you're loading the `*.pvsm` file you'll have to 'load state' [as shown in the slides here](http://wec-sim.github.io/WEC-Sim/_downloads/WEC-Sim_Webinar4.pdf).

stevenshine, 2017-10-04:
hi
1.the wec-sim  code download from https://github.com/WEC-Sim/WEC-Sim_Applications, but it seem to lost some softwave (pressureGlyphs.py and WE-sIM.PY)?
2.i have download  git-lfs but i dont know to use/run it  do u share me more imformations?
which one i can run it ? when i run git-lfs.exe that dont reflect on my PC
[image]
if u dont care,   do u control  my PC  by teamviewer?><
thank u

kmruehl, 2017-10-05:
@stevenshine the ParaView macros are located in the `//WEC-Sim/source/functions/paraview_macros/` directory with the [WEC-Sim source code](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions/paraview_macros). Details on how to install the macros are [included in the visualization webinar](http://wec-sim.github.io/WEC-Sim/webinars.html#webinar-4-mooring-and-visualization) and on slide 26 of the webinar presentation.

Instructions on how to install git-lfs are included on the [git-lfs website](https://git-lfs.github.com/) which is hosted by the developers. This is a github tool, not one developed by the WEC-Sim team. 

Hope this helps.

stevenshine, 2017-10-07:
@kmruehl 
hi
i have finished  install Git  Large file storage , Implemented Git LFS and then  how to downland tracking *.h5 files?
thanks u

kmruehl, 2017-10-10:
1 ) you need to `git pull` to download all the `*.h5` files. 

2/3 ) you need to run the wec-sim model with visualization on to generate all of the paraview visualization files in the `\vtk` directory. Then you can open the ParaView `*.pvd` file or load the `*.pvsm` state.

stevenshine, 2017-10-11:
hi @kmruehl 
there are problem on matlab 
1.i use git lfs download wecsim ,run wecsim in matlab 
whcih i should modify ?
2.where i can download all the *.h5 file?
[image]
[image]
[image]

kmruehl, 2017-10-11:
1. when you installed git-lfs the github repository automatically began tracking the `*.h5` files. They should be in your local folder already. If they are not, you can manually download them from the website - but since you are no longer receiving an error about the `*.h5` file, you should already have them. 
2. I just [resolved this issue with the latest push](https://github.com/WEC-Sim/WEC-Sim). There was an issue with the library block, but it's resolved now. I just ran the ParaView application examples and they are working again.

closed 2017-10-11 by kmruehl

stevenshine, 2017-10-12:
thanks  you 
these quesutons  are solved.
but i still have others questuons,
1.when changing the geometry stl file and OSWE.slx file and then run wecsim  that get error on matlab .
    how to modify?
[image]

[image]
2.

yuyihsiang, 2017-10-12:
From figure 2, you have two constraints, which has the same name.

In addition, it seems that you have floating connection between the flap and the base. For a flapper WEC, the weight of the flap is much smaller than its buoyancy force at its original location. The simulation will most likely crash even it runs, based on what you have.

stevenshine, 2017-10-13:
thanks u a lots  @kmruehl  @yuyihsiang 
these quesutons are solved.