WEC-Sim/WEC-Sim #207: User Defined Wave Imports

MattSharman, 2017-10-02:
Hi All,

I'm looking to run a simulation with some experimental waves and to using the 'etaImport' wave type in WEC-Sim and can't get it to work. 

Am I correct in my definition?
```
% Experimental Waves
waves = waveClass('userDefined');
waves.etaDataFile = 'WaveInputFile_C2Reg_24.txt';
```
With which i get the following message:

> Error using wecSim (line 236)
> Invalid setting in block 'CE_WEC_T2_04/Buoy/Hydrodynamic Body/Nonlinear Wave Elevation/Linear or
> Instantaneous Free Surface/Instantaneous Water Free Surface  /Wave Amplitude' for parameter 'Value'

Could my error be in my .txt file? Does anyone have a working text file i can replicate and test?

Many thanks,
Matt

MattSharman, 2017-10-02:
For the `nlHydro = 0` case i get the following:

> Error using wecSim (line 236)
> Unsupported input format for From Workspace block 'CE_WEC_T2_04/Float/Hydrodynamic Body/Wave Diffraction and
> Excitation Force Calculation/Linear Wave Excitation Force Variant Subsystem/User Defined Wave  Excitation
> Force/InputWaveForceTimeHistory'. Available formats are double non-complex matrix, a structure with or
> without time, or a structure with MATLAB timeseries as leaf nodes. All formats require the data to be finite
> (not Inf or NaN).

MattSharman, 2017-10-02:
After re-watching the non-linear hydro webinar, I realised I can't use user defined waves with nlHydro :sweat_smile:

So I'm now attempting to create my own spectrum to describe my wave state using FFT. I've got my frequencies, amplitudes and phase angles from the FFT and I'm happy with how they describe the wave state:
[image]

But now how do I go about importing this into WEC-Sim as a user defined spectrum?
Many thanks!

kmruehl, 2017-10-02:
@MattSharman you're correct, you _cannot_ run non-linear hydro (`nlHydro`) with user-defined wave time-series (`etaImport`).  Defining the incident wave as a spectrum is the best option to run with non-linear hydro. The wave spectrum should defined in the attached format (also provided in the [RM3 tutorial here](https://github.com/WEC-Sim/WEC-Sim/tree/master/tutorials/RM3)).

[ndbcBuoyData.txt](https://github.com/WEC-Sim/WEC-Sim/files/1349646/ndbcBuoyData.txt)

However, the wave phase is a different story. Is it important for you to model the exact same time-series? If so, you'll need to modify the `waveClass.m` file so that the time-series is reconstructed using the imported wave spectrum and importing a pre-defined phase. Importing a phase is not currently included in the code. The variable you'll need to override is `phaseRand`. 

Let me know if you would like some help on implementing that functionality. If you do modify the code and want to add it to the master version of the code with a pull-request, that would be wonderful.

MattSharman, 2017-10-03:
Many thanks for the help!

Yes, unfortunately it is important to model the exact timeseries. I have a sizeable set of experiments that I'm looking to validate WEC-Sim and my PTO code with and with many of them being non-linear waves (as well as experiencing wave reflection etc.) I'm struggling to achieve a direct comparison with regular wave tests alone.

Some help would be much appreciated as I don't really know where to start. 
I was also wondering whether it could be implemented as a variable condition for MCRs - I have a lot of tests to run through :sweat_smile:. Is that possible or am I asking too much?

kmruehl, 2017-10-03:
@MattSharman I'll look into this and see what I can implement. What is your time-frame?

MattSharman, 2017-10-03:
Short, I've got about two weeks to get this done, project ends in three. 
Many thanks!

kmruehl, 2017-10-10:
@MattSharman we are working on adding this feature and hope to have it pushed by the end of the week.

MattSharman, 2017-10-11:
That's brilliant, many thanks for this! Let me know if you want any help with testing or provide any test cases!

kmruehl, 2017-10-11:
@MattSharman, 
The ability to import wave phase was added with commit **89ed4b3ea7606a44d952dddbf14d6391144211ed**. @yuyihsiang do you have anything you want to add? It looks like the `spectrumDataFile` just needs a third 3 with the phase, correct?

yuyihsiang, 2017-10-11:
It should work now. Please pull the latest version.

1. I have attached a zip file, which has the script that use FFT to convert time history to spectrum and adjust it to the correct amplitude for WEC-Sim to read in. The output file will include three rows of the data (frequency in Hz, spectrum and phase). The zip file also includes an example, which should be straight forward to follow.
[Archive.zip](https://github.com/WEC-Sim/WEC-Sim/files/1377088/Archive.zip)

2. I also updated the waveClass.m file in WEC-Sim, which will check the input spectrum file to see if the third row exist. If it does, it will read it in and use the phase information. In this case, WEC-Sim won't do any interpolation and will directly use the spectrum information within the frequency range given by the BEM solution.

Note: this should work for both nlhydro=0 and nlhydro=2. However, nlhydro=2 will be slow. I planning to update the WEC-SIm lib tomorrow, after Kelley makes a change, which will allow the user (as a default probably) to only calculate the nonlinear restoring and FK forces at every time step but not every iteration.

Let me know if you have any question,

Yi-Hsiang

MattSharman, 2017-10-12:
Thank you very much for this addition, I'll let you know how I get on!
One question: Where do I find the irreg1.txt files referred to in the SpectrumForWECSim.m script?

yuyihsiang, 2017-10-12:
Irreg1.txt is the time history file. If you need one I can send you one after I get in to the office later today.

Equal energy won’t work since the spacing was determined by the input file.

kmruehl, 2017-10-12:
@yuyihsiang Thank you so much for adding this. I'm going to add in a few warning flags to let people know they can't run Equal Energy with direct import. We should also update this on the website...

MattSharman, 2017-10-12:
Yes thank you @yuyihsiang, much appreciated!
Another Q: can I specify an array of `spectrumDataFiles` and get them to run with `wecSimMCR`?

kmruehl, 2017-10-12:
@MattSharman, the short answer is yes. I just [added an application case](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/RM3_MCR/RM3_MCROPT3_SS) for you that does just that. The MCR run specifies the case as a string that corresponds to the data file you want to import. In this cases I specified the third row of data as the phase, and for the purpose of example specified the same phase for all 3 Sea States, see below. 

[image]

yuyihsiang, 2017-10-13:
WEC-SIm is updated. The latest push also includes the speedup on nonlinear hydro calculation.

MattSharman, 2017-10-17:
In the latest update I'm getting an error in wecSimMCR where `waves.randPreDefined` is still called but doesn't exist. Am I right in thinking that it's been replaced by `waves.phaseRand`? I've tried replacing all `randPreDefined` with `phaseRand` but to no avail. :confused:

yuyihsiang, 2017-10-17:
The randPreDefined was changed to phaseSeed. The latest WEC-Sim should work. 
You just need to change you wecSimInputFile and specify waves.phaseSeed = 1,2,3,.....
Sorry for the changes, 

Thanks,

Yi-Hsiang

MattSharman, 2017-10-17:
Sorry, my bad,  I should have been more specific, the latest update isn't working when running regularCIC MCRs.
Up until recently I've been running a collection of regular waves using the `statisticsDataLoad` option and this doesn't work with the latest update. I'm getting this error regardless of `phaseSeed` option:

```
wecSimMCR
No appropriate method, property, or field 'randPreDefined' for class 'waveClass'.

Error in wecSimMCR (line 34)
    if length(waves.randPreDefined)>1;
```

yuyihsiang, 2017-10-17:
Ok I see now.

You can change all the waves.randPreDefined to waves.phaseSeed in the wecSimMCR.m file and that should fix it.
Let me know if it still causing problems

yuyihsiang, 2017-10-17:
Just did a quick check using wecSimMCR and it was fine. Here is part of my inout file

simu = simulationClass();
simu.simMechanicsFile = 'OSWEC.slx';          % Specify Simulink Model File
simu.mode = 'rapid-accelerator';                       % Specify Simulation Mode ('normal','accelerator','rapid-accelerator')
simu.explorer='off';                         % Turn SimMechanics Explorer (on/off)
% simu.startTime = 0;                         % Simulation Start Time [s]
simu.endTime=2000;
simu.solver = 'ode4';                         %simu.solver = 'ode4' for fixed step & simu.solver = 'ode45' for variable step 
simu.dt = 0.05; 							      %Simulation time-step [s] for a convolution function in the radiation force calculation 
simu.rampTime = 250;
simu.CITime = 30;
simu.morrisonElement = 1;

%% Irregular Waves using PM Spectrum with Convolution Integral Calculation
waves = waveClass('irregular');         % Initialize Wave Class and Specify Type
waves.H = 2.25;                          % Significant Wave Height [m]
waves.T = 13*1.16;                            % Peak Period [s]
waves.spectrumType = 'BS';              % Specify Wave Spectrum Type
waves.freqDisc = 'EqualEnergy';
waves.phaseSeed = 1:10;

MattSharman, 2017-10-17:
That's the ticket, thanks!
In order to get this working however I had to comment out my `simu.dtNL=N*simu.dt;` line. Is that expected?

yuyihsiang, 2017-10-17:
No, I don't think you need to comment out that line. It was fine for my test run.
What is the error?

MattSharman, 2017-10-17:
Huh...
> Error using wecSim (line 241)
> Simulink detected an unspecified rate transition at output port 1 of block 'CE_WEC_T2_05/Float/Hydrodynamic Body/Hydrostatic Restoring Force
> Calculation/Linear and Nonlinear Restoring Force Variant Subsystem/Nonlinear Hydrostatic Restoring Force/Add3'. However, Simulink cannot automatically
> insert a Rate Transition block because one of the following scenarios exists:
>  - Fixing the unspecified rate transition requires rewiring of the blocks;
>  - Simulink cannot choose an insertion location for the Rate Transition block because multiple possibilities exist;
>  - The unspecified rate transition requires the insertion of a Rate Transition block on a virtual bus signal.
>  - The output signal of this block is being logged.
>  Rewire the block diagram or manually insert a Rate Transition block to resolve the unspecified rate transition.
> 
> Error in wecSimMCR (line 79)
>     wecSim;

yuyihsiang, 2017-10-17:
I think, for now, you can comment out my simu.dtNL=N*simu.dt. 

I know the error and I have seen that for other block and we have fixed that. However, I could not re-produce this one on my machine right now. Can you send me you case? I should be able to figure it out.

Thanks,

MattSharman, 2017-10-17:
Sure, here you go...
[MSharmanCaseZIP.zip](https://github.com/WEC-Sim/WEC-Sim/files/1391348/MSharmanCaseZIP.zip)

yuyihsiang, 2017-10-17:
Fixed now, just pushed. Let me know if you have any question.

In addition, I would recommend you check to see if simu.dtNL=N*simu.dt gives you the same solution if N>1. I have seen cases, where N>1 gives me different answer when the nonlinear effect is important.

kmruehl, 2017-10-19:
@MattSharman did this resolve the issue?

MattSharman, 2017-10-20:
Yes I believe so, I did get an error in simulink but that was resolved in the latest commit (19th Oct). Many thanks for getting on this so quickly!

kmruehl, 2017-10-20:
glad the issue is resolved :)

closed 2017-10-20 by kmruehl