WEC-Sim/WEC-Sim #208: Running the BEMIO python conversion function and the meshconvert.m file produce two different outputs

rubintaipi1, 2017-10-06:
I have now tried to convert my STL file to a dat file using the original way (via the way described in the BEMIO documentation) and using the meshConvert.m file. They do not produce the same output and neither are an acceptable format that I can use in NEMOH. 

Please help. I am attaching the stl file and the outputs. 
[upperModule.zip](https://github.com/WEC-Sim/WEC-Sim/files/1361296/upperModule.zip)

dforbush2, 2017-10-10:
Hi Rubin,

There is definitely a bug somewhere in the Matlab version. To help me pin it down, if you append everything after the "0 0 0 0" line  from the meshConvert output to the end of the python- produced output, does this allow NEMOH to run for you?  

Thanks. 
Dom

j-vanrij, 2017-10-12:
Hi Rubin,

I’m joining this discussion a little late, but from what I understand - You have a .stl file originating from SolidWorks. You would like to convert this .stl file to a .dat Nemoh input file. However, the recently developed MATLAB mesh conversion capabilities still have some bug to be worked out. 

Although, we fully intend to resolve these MATLAB BEMIO mesh conversion issues, but this could take us some time. So in the meantime, I have a suggestion for you. If you have access to Rhino, or even a free trial version of Rhino, I would highly suggest that you refine your .stl geometry mesh there (since a .stl geometry mesh is generally not great as a BEM input, Nemoh or WAMIT, anyway), save it as a WAMIT .gdf mesh, then use either the old python BEMIO, or new MATLAB BEMIO to convert the .gdf to NEMOH .dat file. I realize this creates an extra step for you, but it will likely get you going sooner than waiting for us!

Thanks for the questions (and debugging!)

Jennifer

dforbush2, 2017-10-18:
Hi Rubin,

As an update on our end, I was able to run NEMOH with your mesh, but I think replicated the issues you were having with meshConvert.m: a large number of panels were improperly defined in the connectivity table (the second part of the .dat file, below the '0 0 0 0' line) such that they were either open, or their normal vector was inward-facing. This triggers a warning in NEMOH that "Normal vector of panel #### is in the X direction". Right now, the nodes are connected in the order in which they are read in to the file. While this apparently worked on my initial test files, it clearly does not work in general (it may not work in general gdf->dat either). As Jennifer stated, this is not a trivial issue and will likely take me awhile to fix. I immensely appreciate your help in finding this problem. 

Were you able to find a way to make your model work?

In addition to Jennifer's suggestions, I would recommend checking out the mesh utilities recommended by NEMOH developers. Mesh.m and the MeshMagick utility (in python) may be a work-around that will give you success. MeshMagick (https://lheea.github.io/meshmagick/command_line/cli_usage.html) offers a higher level of functionality as well that you may find useful. Further, if you are able to make a GDF equivalent mesh, there are two functions that allow NEMOH to run using a GDF mesh (https://lheea.ec-nantes.fr/logiciels-et-brevets/nemoh-mesh-192932.kjsp?RH=1489593406974). 

Thank you for your efforts here, and I am sorry I don't have a quick solution for you,
Dom

dforbush2, 2017-10-31:
Hi Rubin,

I was checking in on your status with regard to this issue. I have done some investigations into meshMagick using your mesh, and unfortunately, it seems that this utility produces a similar output to the other tools, resulting in an identical error when run in NEMOH. Moreover, the panel size on the vertically-oriented faces of your mesh is likely too large to generate reliable BEM results. 

As a final suggestion, your mesh appears as though it could be described fairly easily by a series of equations. With this in mind, it may be possible to manually construct a dat file of the correct format/normal orientation using built-in matlab functions (see mesh, trimesh). 

Hope all is well,
Dom

closed 2017-11-14 by kmruehl