WEC-Sim/WEC-Sim #211: Hydrodynamic software/CFD in examples

waveenergyharnessingdevice, 2017-11-04:
Both of the tutorial examples in WEC-Sim use WAMIT. To be fair to developers of other hydrodynamic software products, such as AQWA and NEMOH, example runs of each should be provided

kmruehl, 2017-11-06:
Thank you for this feedback. We have provided tutorials using BEM data generated from AQWA, NEMOH and WAMIT on the [wec-sim website here](http://wec-sim.github.io/WEC-Sim/advanced_features.html#bemio), and within the [BEMIO tutorials included in the source code here](https://github.com/WEC-Sim/WEC-Sim/tree/master/tutorials/BEMIO). 

If you're interested in running the RM3 or OSWEC tutorial with another BEM code, you can do so running of of these BEMIO tutorials and copying the generated `*.h5` file into the `\hydroData` directory for the desired run.

sugarl14, 2020-01-22:
Hey Kelley,

I know this is a pretty old post, but I have a couple questions about it:
1) Are there any new links that you are aware of that answer @waveenergyharnessingdevice 's question (particularly using AQWA)?
2) if not, I believe the second hyperlink you posted is broken. Is there any way you could send me an updated link?

Thank you!
Landon

kmruehl, 2020-01-22:
We provide examples running BEMIO for WAMIT, AQWA and NEMOH. 

1. The BEMIO documentation is available here: http://wec-sim.github.io/WEC-Sim/advanced_features.html#bemio-functions 
2. The examples using BEMIO for WAMIT, AQWA and NEMOH arehere: https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/BEMIO

closed 2020-01-22 by kmruehl