WEC-Sim/WEC-Sim #212: problem while installing

mathyturing, 2017-11-06:
i'm new in wec-sim, i do not suceed in installin wec-sim
how to find matlab startup folder?
when I type pwd on the matlab command, I get a directory but there is no startup .m file there
should I create one?
I have created one and copied the code contained in the wecSimStartup.m
I have also change the value of the variable wecSimPath as follows
[image]

but i get the error 
startup.m not found
when I write open startup.m
on matlab command window

kmruehl, 2017-11-06:
@mathyturing your `startup.m` should be located in your MATLAB path. Typically, this is `C:\Users\yourUserName\Documents\MATLAB`. If there is not a `startup.m` file in that directory, add the one you created. 

FYI `startup.m` should be located in the MATLAB path (not the WEC-Sim directory).

closed 2017-11-06 by kmruehl

mathyturing, 2017-11-07:
thanks for your quick reaction.

mathyturing, 2017-11-07:
and what should be the value of the wecSimPath variable?
I have set it to the path to the folder into wich the WecSimStartup is contained.
I have done the following
[image]

are  they the right values??/

kmruehl, 2017-11-07:
That is the correct path. The WEC-Sim path refers to the location of the `WEC-Sim/source` directory. 

[see online doc here](http://wec-sim.github.io/WEC-Sim/getting_started.html#step-1-add-wec-sim-source-code-to-matlab-path)

mathyturing, 2017-11-08:
thank U very much