WEC-Sim/WEC-Sim #214: RM3_cHydraulic_PTO: Getting errors when running irregular waves

ialci, 2017-11-14:
Hello,
I was trying to run the tutorial case of RM3_cHydraulic_PTO without making any modifications and I got the following errors:
Error in wecSimInputFile (line 16)
waves.randPreDefined=1;

Error in wecSim (line 29)
evalc('wecSimInputFile');

So I checked the waveClass.m file and there is nothing about randPreDefined. Then, I tried taking out the sentence:
waves.randPreDefined=1; (line16 of the wecSimInputFile)

However, I got different errors:

Error using error_for_persistent_vars_in_ct_mlfcn (line 9)
MATLAB Function uses constructs that are invalid when the block specifies or inherits a continuous sample time.
Invalid constructs include the use of persistent or global variables, calls to exported functions or using
coder.extrinstic.

Error in sfprivate (line 15)
       feval(fcnName,varargin{:});

Error in wecSim (line 241)
sim(simu.simMechanicsFile);

If I switch to regular waves, the simulation works! It seems like it is an issue with irregular waves. Could you please check?
Thank you very much!

kmruehl, 2017-11-14:
@ialci the issue is that the variable named `waves.randPreDefined` was changed to `waves.phaseSeed` and I forgot to update it in the input file. If you change the line to `waves.phaseSeed=1;`, the simulation should run without error. I will also update the input file on the WEC-Sim_Applications repository to reflect this.

ialci, 2017-11-15:
Thank you very much for the quick response!

However, I still get errors: 

Error using error_for_persistent_vars_in_ct_mlfcn (line 9)
MATLAB Function uses constructs that are invalid when the block specifies or inherits a continuous sample time.
Invalid constructs include the use of persistent or global variables, calls to exported functions or using
coder.extrinstic.
Error in sfprivate (line 15)
feval(fcnName,varargin{:});
Error in wecSim (line 241)
sim(simu.simMechanicsFile);

I tried to run the RM3 tutorial using different types of waves (without changing anything: just removing %) and the error is the same: 'etaImport' works well but neither 'irregular' nor 'spectrumImport' work.

Best regards

ialci, 2017-11-15:
@kmruehl Thank you very much for the quick response!
However, I still get errors:
Error using error_for_persistent_vars_in_ct_mlfcn (line 9)
MATLAB Function uses constructs that are invalid when the block specifies or inherits a continuous sample time.
Invalid constructs include the use of persistent or global variables, calls to exported functions or using
coder.extrinstic.
Error in sfprivate (line 15)
feval(fcnName,varargin{:});
Error in wecSim (line 241)
sim(simu.simMechanicsFile);
I tried to run the RM3 tutorial using different types of waves (without changing anything: just removing %) and the error is the same: 'etaImport' works well but neither 'irregular' nor 'spectrumImport' work.
Best regards

kmruehl, 2017-11-15:
@ialci, two questions:
- What version of matlab are you running?
- Would  you be able to ZIP your model and upload it here? I am unable to reproduce this error on my end, and will need to see the model you're running to debug. 

Thanks!

ialci, 2017-11-15:
@kmruehl , thank you for your assistance!

I'm running MATLAB and Simulink R2017b for Windows 10 x64bit.

I atach the slx and m input files. (I just took them from: 'WEC-Sim_Applications/PTO-Sim/RM3_Hydraulic_PTO/' and I run them without any modification--just changing waves.randPreDefined as you told me)

Thanks!
[forKelley.zip](https://github.com/WEC-Sim/WEC-Sim/files/1476374/forKelley.zip)

kmruehl, 2017-11-29:
@ialci Thank you for providing your model. 

The issue may have been due to the MATLAB 2017b version. We just did some updates to the WEC-Sim code. With these updates I was able to run the model you provided, and the [RM3_cHydraulic_PTO ](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/PTO-Sim/RM3_cHydraulic_PTO) model on my Windows 10 x64bit machine with MATLAB 2017b. There were other people who previously ran into issues running WEC-Sim with 2017b as well, but these issues have been resolved with the latest WEC-Sim updates. Please pull the [latest version of WEC-Sim master](https://github.com/WEC-Sim/WEC-Sim). This should resolve your issue. 

Thanks!

closed 2017-11-29 by kmruehl

ialci, 2017-12-02:
Thank you very much! It works perfectly now!

commit 8eeb8499 (2017-11-14): updated submodule, resolved #214