WEC-Sim/WEC-Sim #216: RM3 with hydraulic PTO (instalation)

JPMS-Chile, 2017-11-16:
Hello! i'm new here and i'm trying to use the RM3 with hydraulic PTO model.

I downloaded the model from GitHub, but there are some blocks are missing in the PTO library (compared with https://wec-sim.github.io/WEC-Sim/advanced_features.html). Specifically "Translational PTO UD Force (Local Z)".

I want to study the behavior of an power converter conected to the chilean electrical grid for my final exam in electrical engieenering, so, my idea was continue the model from the generator to the grid, though the power converter, bateries, lines, others.

I will appreciate any help or recommendation (and sorry for my english)

yuyihsiang, 2017-11-28:
Please make sure you have the latest version of WEC-Sim.

I have tested the case and it seems working fine (on my machine). It will help if you can take a snapshot on the error if you still have the same issue after pulling the latest WEC-Sim version.

Yi-Hsiang

closed 2017-11-28 by kmruehl

rohitalexander, 2021-07-16:
Hello Team!

I am working on a project for my M.Sc dissertation. I was looking into if I can implement PTOSim for my project. While I was trying out the RM3 tutorial using PTOSim, I noticed that I am having the same issue mentioned here. The PTO Library does not contain the "Translational PTO UD Force (Local Z)" block. I think I am using the latest WEC-Sim version (I downloaded the source code from : https://github.com/WEC-Sim/WEC-Sim). 

Please let me know how this can be resolved.

Many Thanks,
Rohit
 
[image]