WEC-Sim/WEC-Sim #217: Problem simulating an OWSC

NMartinez9, 2017-11-16:
Hello everyone,

I am working on a OWSC, just like the one in WECSim's Tutorial site, but the dimensions are differet. The flap is 1,31m*0.17m*0.84m (wide*thickness*high) and the base the same dimensions but 0.10m high.

For the hydrodynamic data i'm working with Nemoh so, i did mesh separately the flap and the base, and ran the code for both, generating one Nemoh.cal file.

Following the example in WECSim's Tutorial, i created the .h5 file using bemio.m. Using Autocad 2016 i drew the same structures and exported as .stl files, with the CoG on the origin of coordinates.

The wecSimInputFile.m is similar to the OWSC example, but changing the mass, the moment of inertia (about the CoG, am i rigth?) and the location of the rotational PTO (at the end of the flap).

So far, apparently no problem.

When i try to simulate for regular waves, T=3.5 and H=0.25, it gave an error "Derivative of state 'P2.Rotational_PTO.Pitch_PCC_Revolute_Joint.Rz.w' in block 'P2/Rotational_PTO/Pitch_PCC (Revolute Joint)' at time 0.305 is not finite. The simulation will be stopped. There may be a singularity in the solution"

I also noticed that in the Mechanics Explorer the frame that should be in the CoG for the flap appears in its lower corner. (When i ran the OWSC example, the frame was at CoG, can be a problem whit the .stl file?)

I'm not sure if this is a hydrodynamic problem and/or the simulink model does not work. I've been looking in the forum, so i try to change the simu.dt to T/200 but does not work. I also try changing the T but allways seem to fail. 
I read another post that suggests to calculate the RAO, and wecSim can do so, but i do not know how to do that.

Any suggestions on how I can resolve this error? Thank you in advance, hope I have been clear in describing the problem.

Regards

yuyihsiang, 2017-11-28:
"The frame that should be in the CoG for the flap appears in its lower corner"

I am wondering if you have the correct coordinate system in Nemoh. For WEC-Sim, the CG of each body is directly obtained from Nemoh. I would recommend you to check that.

Yi-Hsiang

NMartinez9, 2017-11-28:
Hello Yi-Hsiang, thank for your reply.

I am using the matlab routine Mesh.m for meshing the body; look like this

nBodies=1;
n=5;
X(1,1,:,:)=[0.085 0. -0.84;-0.085 0. -0.84;-0.085 0.655 -0.84;0.085 0.655 -0.84];
X(1,2,:,:)=[0.085 0. -0.84;0.085 0.655 -0.84;0.085 0.655 0;0.085 0. 0]; 
X(1,3,:,:)=[0.085 0.655 -0.84;-0.085 0.655 -0.84;-0.085 0.655 0;0.085 0.655 0];
X(1,4,:,:)=[-0.085 0. 0;0.085 0. 0;0.085 0.655 0;-0.085 0.655 0];
X(1,5,:,:)=[-0.085 0.655 0;-0.085 0.655 -0.84;-0.085 0. -0.84;-0.085 0. 0];
tX(1)=0.;
CG(1,:)=[0. 0. -0.51];
nfobj(1)=1000;
[Mass,Inertia,KH,XB,YB,ZB]=Mesh(nBodies,n,X,tX,CG,nfobj);

I believe that is the right way to define the flap, considering that this matlab routine take account of symmetry on xOz. Nevertheless makes me wonder if is necessary to define the top panel below z=0 because WEC-Sim will be use those coefficients for calculation.

I tried with .STL geometry file from Salome open source instead of the .STL file from Autocad , and when run WEC-Sim it seem to show CoG in the right place. I understand that for linear cases WEC-Sim only need .STL files for showing on Mechanics Explorer, is it? If it is, beyond the fact that the gravity center is not shown in the right place, the calculation should be correct?

Anyway, i still having the error described on the first post and i don't understand why. 
On **OSWEC: Regular Wave Case #41** suggest that can be an instability that can be check with the RAO calculation. How can i do that? 

Is there something I'm missing?

Thank you very much for your help.

Regards

yuyihsiang, 2017-11-28:
Yes, the STL file does not affect the results. Given that you get the CoG showing correctly using Salome, you should be fine with the coordinate system. 

Here is my list of things you may want to check.
1. pto.loc
2. body moment of inertia
3. the CoG and location for forces calculation in Nemoh.cal 

Yi-Hsiang

NMartinez9, 2017-11-29:
Hello again,

1) For the simplified model, I located the Rotational PTO and Fixed constraint as shown below

[image]

And for the wecSimInputFile the next characteristics, because the PTO is located in the hinge attached to the channel floor by means of the fixed constraint.

%% PTO and Constraint Parameters

% Fixed
constraint(1)= constraintClass('Constraint1');  % Initialize ConstraintClass for Constraint1
constraint(1).loc = [0 0 -0.85];                  % Constraint Location [m]

% Rotational PTO
pto(1) = ptoClass('PTO1');                      % Initialize ptoClass for PTO1
pto(1).k = 0;                                   % PTO Stiffness Coeff [Nm/rad]
pto(1).c = 0;                                   % PTO Damping Coeff [Nsm/rad]
pto(1).loc = [0 0 -0.85];                        % PTO Location [m]

2) The body moment of inertia is 40.72 kg.m2 about the hinge and 32.85 kg.m2 about CoG. I read on WEC_Sim_User_Manual_v1.0 that for the wecSimInputFile it must be given about CoG, it is correct?

3) On the same manual tells that all forces and moments has to b given at CoG, and for the Nemoh.cal file, software automatically put the force and moment on CoG. It is correct, or for this case, i must correct the point?
This is the Nemoh.cal file

mesh\mesh1.dat      ! Name of mesh file
2041 1836			! Number of points and number of panels 	
6				! Number of degrees of freedom
1 1. 0.	0. 0. 0. 0.		! Surge
1 0. 1.	0. 0. 0. 0.		! Sway
1 0. 0. 1. 0. 0. 0.		! Heave
2 1. 0. 0. 0.000000 0.000000 -0.520000		! Roll about a point
2 0. 1. 0. 0.000000 0.000000 -0.520000		! Pitch about a point
2 0. 0. 1. 0.000000 0.000000 -0.520000		! Yaw about a point
6				! Number of resulting generalised forces
1 1. 0.	0. 0. 0. 0.		! Force in x direction
1 0. 1.	0. 0. 0. 0.		! Force in y direction
1 0. 0. 1. 0. 0. 0.		! Force in z direction
2 1. 0. 0. 0.000000 0.000000 -0.520000		! Moment force in x direction about a point
2 0. 1. 0. 0.000000 0.000000 -0.520000		! Moment force in y direction about a point
2 0. 0. 1. 0.000000 0.000000 -0.520000		! Moment force in z direction about a point
0				! Number of lines of additional information 

Thank you

yuyihsiang, 2017-11-29:
Seems all right. 

We can take a look on your model if you can upload the files (zip them).

Thanks,

Yi-Hsiang

NMartinez9, 2017-11-29:
Hello Yi-Hsiang,

I attached the zip file. Let me know if there is some information missing.

[Model.zip](https://github.com/WEC-Sim/WEC-Sim/files/1515742/Model.zip)

Thanks!

yuyihsiang, 2017-11-29:
I tested it. Please change simu.dt to 0.01 and it works on my machine.

Thanks,

Yi-Hsiang

NMartinez9, 2017-11-30:
Hello Yi-Hsiang, 

It is working now. Thank you for the help! 

Regards

closed 2017-11-30 by kmruehl