WEC-Sim/WEC-Sim #218: Problem with BEMIO - WAMIT & generalized modes

enricoande, 2017-11-17:
I have been trying to approximate the radiation force with a state-space system using BEMIO. 5 DOF are turned on right now (1,3,5 + 2 generalized DOFS), but BEMIO read my WAMIT files as though it had 6.
I was wondering if it is the generalized modes that are causing the problem and what possible alternatives would be?
Thank you!

j-vanrij, 2017-11-28:
Currently, all of the BEMIO read/write functions assume that each body has 6 degrees of freedom. Read_WAMIT (but not Read_NEMOH or Read_AQWA) was modified, but not extensively tested, to also read in generalized body modes (if the output directory includes *.cfg, *.mmx, and *.hst files).

If the .gdf file was run with symmetries, Read_WAMIT should still work, but all of the results will still be written out assuming 6 dof. If it does not, feel free to attach your WAMIT input/output files here and we can take a look at them.

And, just to be forewarned, if you are just using BEMIO to approximate the radiation force with generalized modes, that should work, but the current, public version of WEC-Sim cannot handle generalized modes. 

Thanks for the question.

closed 2017-11-28 by kmruehl

enricoande, 2017-11-29:
Thank you very much for that!