WEC-Sim/WEC-Sim #219: Error Running RM3 MoorDyn

blazingdog2, 2017-11-17:
Hi,

I have downloaded C/C++ and the moordyn files as mentioned in the webinar. However every time the simulation is being ran the following error is shown.

Error using wecSim (line 241)
Error in 'RM3MoorDyn/MoorDyn': Initialization commands cannot be evaluated.
Caused by:
    Error using textread (line 159)
    File not found.

[image]

Please assist, 
Thank you

yuyihsiang, 2017-11-28:
You are missing the mooring folder and the moorDyn setup file (lines.txt) inside the folder.

Please pull the latest WEC-Sim (or directly from WEC-Sim_Applications repo), which should have a mooring folder in this example folder.

Yi-Hsiang

blazingdog2, 2017-12-01:
Hi sorry, after placing the moordyn and mooring folder i am still getting the same error.

[image]

Wei Jie

kmruehl, 2017-12-01:
@blazingdog2 did you run `wecSim` from the local path of your`wecSimInputFile.m`? In the screenshot you've included, your MATLAB path is not `\RM3_MoorDyn`.

If you still have this error, please ZIP your model and @yuyihsiang will help debug. Thank you.

blazingdog2, 2017-12-03:
Its working now thank you very much for your help!

Wei Jie

closed 2017-12-03 by blazingdog2

Legaze, 2018-12-05:
Hello,
I am running the same tutorial but I am getting a different problem with the software, could you help me, I asked the same question in the wrong issue; sorry about it.

[image]

Thanks a lot,
Leo