WEC-Sim/WEC-Sim #220: UD Rotational PTO force sign inconsistencies

bradling, 2017-12-01:
It looks like there is an inconsistency in the sign of the input and output PTO forces for the Rotational PTO Actuation Torque block.

The values input to the block have the opposite sign as values in `output.ptos(1).forceActuation(:,5)`

yuyihsiang, 2017-12-01:
We are trying to figure out if we need to re-tag the v3.0 or not. Have you seen similar issue in the past or this is something new you just find out. 

Thanks,

Yi-Hsiang

bradling, 2017-12-01:
I think it has always been this way, and I have simply adapted to the convention used. I am working on a different WEC-Sim model with some folks at OSU and the inconsistency created some problems.

bradling, 2017-12-01:
Here is a screenshot of the block. I think it arises from the coordinate transformation of the Pitch PCC joint block. If you look closely, youll notice that there is a -1 gain on all the pitch component output forces (circled in red), but no -1 gain on the input torque (circled in blue).

It looks like this convention is also inconsistent with the linear PTO actuation torque block.

[image]

yuyihsiang, 2017-12-04:
Ok, thanks, Brad. I'll take a look today or tomorrow.

Thanks,

Yi-Hsiang

yuyihsiang, 2017-12-08:
Hi Brad

The gain was specified to correct the direction for one of the rotational direction due to the coordinate transformation. As you have said, It is only required for rotational PTOs not the translational one. However, in order to make the actuation force consistent with the coordinate system, the block is modified to address the issue. Please see the attachment for changes

[Update-to-PTO-Rotatry-Actuation-Block.pdf](https://github.com/WEC-Sim/WEC-Sim/files/1543687/Update-to-PTO-Rotatry-Actuation-Block.pdf)

I'll close this issue but let us know if you have any question.

Yi-Hsiang

closed 2017-12-08 by yuyihsiang

bradling, 2017-12-09:
Thanks! And thanks for the detailed documentation of the changes. Looks to me like that should fix the issue.

yuyihsiang, 2017-12-10:
Nathan did the work and documentation. 

Thanks,

Yi-Hsiang