WEC-Sim/WEC-Sim #221: Translational PTO Actuation force

Krishnendu-Ocean, 2017-12-01:
Hello,
Please help me to solve an issue regarding translational PTO actuation force.

I have introduced a translational PTO actuation force between two bodies; one is a float and other is a non-hydro body (Fixed). The movement of the float is restricted by the second body along Z direction (heave). I am not able understand where should I connect the force and response in the translational PTO actuation force block. I tried introducing a PTO-sim block as a trial but I am not able to get the output. I wanted the position of the float at each time step as output. Please let me know how I can introduce these parameters in my model and do I need to do any changes in the WEC-Sim code. Please note that I am trying to simulate model response without any PTO system. 

Thank you.

kmruehl, 2017-12-01:
@Krishnendu-Ocean Can you ZIP your model and upload it here? We can help debug that way. Thanks.

Krishnendu-Ocean, 2017-12-03:
Thanks for your reply.
I am attaching the simulink model and a picture of my experiment model testing for your reference. 
I am concerned about the displacement of the plate model with the restriction of the rod (non-hydro body) which is rigidly fixed. Thank you in advance. 

[Plate model-Expt-Photograph.zip](https://github.com/WEC-Sim/WEC-Sim/files/1524728/Plate.model-Expt-Photograph.zip)

[Plate model.zip](https://github.com/WEC-Sim/WEC-Sim/files/1524727/Plate.model.zip)

nathanmtom, 2017-12-12:
@Krishnendu-Ocean If you are only trying to simulate the model without any PTO system you do not need to use a PTO block. All you need to use is a Translational Constraint Block. This will constrain the motion of the WEC to heave only and you will not need to worry about connect the response output or force input in the PTO block that you provided.

nathanmtom, 2017-12-19:
@Krishnendu-Ocean Did this solve your problem? We would like to close this issue.

closed 2017-12-20 by kmruehl