WEC-Sim/WEC-Sim #225: MATLAB 2017b specific error

bretbosma, 2017-12-19:
Hello,

Model runs fine on 2017a but gives the following in 2017b.  Any thoughts?

Thanks,
Bret

Error using error_for_persistent_vars_in_ct_mlfcn (line 9)
MATLAB Function uses constructs that are invalid when the block specifies or inherits a continuous sample
time. Invalid constructs include the use of persistent or global variables, calls to exported functions or
using coder.extrinstic.

Error in sfprivate (line 15)
       feval(fcnName,varargin{:});

Error in wecSim (line 241)
sim(simu.simMechanicsFile);
Suggested Actions:
    • Set sample rate of this block to inherit the fastest discrete rate in the model. - Fix
    • Open the properties dialog of this block to set the sample time manually. - Open

yuyihsiang, 2017-12-19:
Hi Bret

Are you using the latest WEC-Sim?
I believe we have fixed that issue in the latest WEC-Sim version. If you have to use the older version, you should be able to just comment out (or delete) that specific line in the code (that's how we have fixed it). It was not used anyway. Let us know if that fixes you issue.

Thanks,

Yi-Hsiang

bretbosma, 2017-12-19:
Hi Yi-Hsiang,

I am using the latest WEC-Sim, downloaded yesterday.  Which line do you recommend commenting out?

Thanks,
Bret

yuyihsiang, 2017-12-20:
Hi Bret

According to the error, it is related to the use of persistent variables in one of the MATLAB function block. However, I was not able to reproduce your error. 
Did you create any block yourself? Can you send me you model? I can try it on my machine.

Thanks,

Yi-Hsiang

bretbosma, 2017-12-20:
Hi Yi-Hsiang,

Thanks for the help.  I found a solution.  I had to replace the Hydrodynamic body blocks with new ones and it worked.

Regards,
Bret

closed 2017-12-20 by bretbosma