WEC-Sim/WEC-Sim #226: Question about selector block

Felipebalboa, 2017-12-24:
Hi, thanks for the WEC-Sim tool.

I have very little time interacting with the code in Matlab (Matlab2015b), recently started to make the diagram in Simulink and I have a question regarding the selector block. 

I am studying an OWSC device, therefore, a rotational motion which should represent the selector block as shown in figure will have.

[image]

But, I do not understand how to configure the parameters of the selector block to that configuration.

[image]

I would appreciate if I could help with the step by step to understand the operation of the block; It also would make progress in my study.

Thank and happy celebration!

yuyihsiang, 2017-12-29:
Hi Balboaf 

The index you showed in the second figure means that you will output the first and third value of the input vector. In other words,

output vector has a size of 2 and will equal to [v(1) V(3)], where v is the input vector.

More details can be found in
https://www.mathworks.com/help/simulink/slref/selector.html?requestedDomain=www.mathworks.com

Hopefully, this is helpful for you.

Yi-Hsiang

balboaf, 2018-01-02:
Then, for the configuration shown in the first figure must exit the fifth value of the input vector. The vector of output will be of size 1 and would be equal to the vector of input v (5). Would the size of the value of "Input Port Size" is 6?

[image]

kmruehl, 2018-01-04:
Correct, `input port size` refers to the number of input ports, in this case 6 for the number of DOF input. And `index` refers to the index number of the input signal, in this case 5 for pitch (RY) motion.

closed 2018-01-04 by kmruehl