WEC-Sim/WEC-Sim #227: Missing WEC-Sim blocks in Simulink

gui-paredes, 2018-01-03:
I have installed WEC-Sim V3.01 from the zip file available at https://github.com/WEC-Sim/WEC-Sim/releases, according to the instructions in https://wec-sim.github.io/WEC-Sim/getting_started.html.

After installation and refreshing the simulink library, the WEC-Sim sub-library is empty, while the PTO-Sim sub-library displays its blocks.

I am using ubuntu 16.04 with MATLAB Version 9.3 (R2017b), Simulink Version 9.0 (R2017b), Simscape Version 4.3 (R2017b), and Simscape Multibody Version 5.1 (R2017b).

How can I fix this?

Thank you,
Best regards,
Guilherme Paredes

[image]

kmruehl, 2018-01-04:
@gui-paredes thanks for this question. The issue is that the v3.01 tagged release is missing the WEC-Sim library blocks (a mistake on our part). We are working on resolving this issue in the tagged release. In the meantime, please download the ZIP file from the [WEC-Sim master branch](https://github.com/WEC-Sim/WEC-Sim). This is down by clicking the green `Clone or download` button and selecting `Download ZIP` as shown in the screenshot below. 

[image]

gui-paredes, 2018-01-12:
Thank you for you assistance, @kmruehl

closed 2018-01-17 by kmruehl