WEC-Sim/WEC-Sim #228: Missed rampT in Morrison

Toby-D, 2018-01-06:
WEC-Sim Team, 

v3.0 looks good! Also, coming back to WEC-Sim after being away for several months, the documentation is looking great. I think that's really important, so thanks for updating that. 

Here's an easy bug for you: One block seems to have been missed when updating variable names:
"WECSim_Lib/Body Elements/Rigid Body/Hydrodynamic Body/Morrison Element and ViscousDampingForceCalculation/Morrison Element Variant Subsystem/Morrison Element On/No Wave Morison Force/MorrisonElements/OrbitalVelocity" needs "simu.rampT" to be replaced with "simu.rampTime".

As an aside, it's not obvious to me why that Morison Element has a simulink version of OrbitalVelocity while the others use the .m function. But maybe there's a reason.

Toby-D, 2018-01-06:
Also, that subsystem calls waves.k(1,:), but waves.k appears to be empty when running noWave simulations.

kmruehl, 2018-01-08:
@Toby-D Thanks for the feedback on the documentation, we've been working on improving it, and I'm glad to hear it's beneficial. Thanks for catching this bug, I resolved the `simu.rampT` bug in the latest commit. 

As for the question as to why the orbital velocity is calculated in Simulink instead of using a *.m function, I believe this is a legacy of guidance we received from Mathworks a long time ago that it's faster to run Simulink blocks than to run the MATLAB function block. However, that is no longer true, and we've worked to transition to function blocks. I believe the Morison Element is simply a case we haven't gotten around to changing. I made note of this on the [project page](https://github.com/WEC-Sim/WEC-Sim/projects/16). 

Also, when you run the Morison Element with no wave, does the simulation crash? The wave number should be calculated based on the `waves.T` defined  in the input file. If not, we can look into resolving this issue.

Toby-D, 2018-01-09:
Thanks Kelley.

Yes, WEC-Sim crashes when you include Morison elements with no waves. waveClass (line 116) doesn't call wave.waveNumber for noWave runs, but the no-wave Morison block looks for the wave number in the orbital velocity block.

An easy workaround is to just use regular waves with waves.H=0.

Toby-D, 2018-01-09:
PS Sorry I didn't point this one out separately--besides the constant input block that you changed, there's also an instance of "rampT" in WECSim_Lib_debugged/Body Elements/Rigid Body/Hydrodynamic Body/Morrison Element and ViscousDampingForceCalculation/Morrison Element Variant Subsystem/Morrison Element On/No Wave Morison Force/MorrisonElements/OrbitalVelocity/Applied Ramp Function.

kmruehl, 2018-01-17:
@Toby-D this latest push should resolve all of these issues. Thank you very much for identifying them and providing a suggested resolution. FYI, we are retagging v3.0 to include these revisions.

closed 2018-01-17 by kmruehl

Toby-D, 2018-01-17:
Great. Thanks Kelley!

<https://github.com/WEC-Sim/WEC-Sim/issues/228#issuecomment-358461347>, or mute
the thread

commit 27cfb3b6 (2018-01-17): resolved #228, updated ME blocks and added error flag