WEC-Sim/WEC-Sim #229: Run WecSIM+MoorDyn on Linux

gui-paredes, 2018-01-25:
I am trying to run WecSIM with MoorDyn on Ubuntu Linux, but I get the following error:
 

> Cannot run MoorDyn in this platform
Error evaluating 'InitFcn' callback of SubSystem block 'RM3MoorDyn/MoorDyn/MoorDyn/MoorDyn'.
Callback string is '% Initialize MoorDyn Lib (Windows:dll or OSX:dylib)
if ismac
   loadlibrary('Lines.dylib','MoorDyn.h');
else
   if ispc
      loadlibrary('Lines.dll','MoorDyn.h');
   else
      disp('Cannot run MoorDyn in this platform');
   end
end
calllib('Lines','LinesInit', mooring.loc ,[0 0 0 0 0 0]);'
Caused by:
    Library was not found

I am using the MoorDyn distribution available on github (https://github.com/WEC-Sim/moorDyn), after reading issue #104. The problem seems to be that, even though there is a library for linux (Lines.so), WecSIM will stop even if the OS is Linux. I tried to edit the function initFcn in the MoorDyn block, to add an option for linux, but I couldn't. How can I fix this issue?
Thank you,
Best regards,
Guilherme Paredes

gui-paredes, 2018-01-26:
I think I figured out how to make it work, but I would like to get some feedback from the developers on my solution, if possible.
Here it goes:
1) since I couldn't edit the Simulink block functions directly in Simulink, I used an xml editor to edit the file blockdiagram.xml under source/lib/WEC-Sim/WECSim_Lib.slx/simulink. I replaced lines 24663 to 24671 with 
`if ismac`
`loadlibrary('Lines.dylib','MoorDyn.h');`
`elseif ispc`
`loadlibrary('Lines.dll','MoorDyn.h');`
`elseif isunix`
`loadlibrary('Lines.so','MoorDyn.h');`
`else`
`disp('Cannot run MoorDyn in this platform');`
`end`

This will allow WEC-Sim to start MoorDyn in a unix-like OS, provided the file Lines.so is present in the MoorDyn code folder.

2) Because Linux is case sensitive, I had to change, in the simulation case folder, the folder name where the file 'lines.txt' is located from 'mooring' with lowercase m to 'Mooring' with upper case M. Apparently, MoorDyn looks for a folder with an uppercase M, which does matter in Linux, even if it doesn't in windows.

3) Then I had to change all internal references in the WEC-Sim source code to the folder 'mooring' with lower case m to 'Mooring' with uppercase M. I did this in source/objects/mooringClass.m on line 88, and in the source/objects/responseClass.m on line 94.

yuyihsiang, 2018-01-31:
Hi Guilherme

Yes, somehow the linux option got dropped out, which I don't know why.

I put it back, as exactly the way you did. So the latest version should work now. 

However, I don't have Matlab on my our Linux server. It will be great if you can check it and let us know.

Thanks,

Yi-Hsiang

commit ed0468d2 (2018-01-31): Fixed #229 unable to run in Linux issue
closed 2018-01-31 by yuyihsiang