WEC-Sim/WEC-Sim #230: Error when running WEC-Sim

genius613, 2018-02-07:
When I run WEC-Sim, I get the following error in MATLAB:

WEC-Sim Read Input File ...   
Undefined variable "waves" or class "waves.checkinputs".

Error in wecSim (line 41)
waves.checkinputs;

I was wondering if anybody could help me with this.

kmruehl, 2018-02-07:
It sounds like your `wecSimInputFile.m` does not properly initialize the waveClass. Refer to the [waveclass section](http://wec-sim.github.io/WEC-Sim/code_structure.html#wave-class) in our documentation and our [RM3 tutorial](https://github.com/WEC-Sim/WEC-Sim/blob/master/tutorials/RM3/wecSimInputFile.m) for examples of how to initialize the wave class and define WEC-Sim's wave input.

genius613, 2018-02-07:
I should mention that I am trying to run the RM3 and OSWEC tutorials provided by the WEC-Sim developers and have not adjusted the code in the wecSimInputFile.m file or any other file.

kmruehl, 2018-02-07:
Both of those tutorials should run without modification. Are you running them from the case directory, and have you added the wec-sim source directory to your path?
http://wec-sim.github.io/WEC-Sim/overview.html#running-wec-sim 

If so, please provide a screenshot of your error.

genius613, 2018-02-09:
I have had a lot of trouble adding the wec-sim source directory to my path. I found the instructions on the website to be extremely confusing so I ended up somehow doing it manually, but this means I have to add it each time I open MATLAB which is a big hassle. Do you have a way to avoid having to do this?

For the case directory, do I use the case directory found in the downloaded wec-sim folder or do I need to create my own?

kmruehl, 2018-02-12:
The easiest way to automatically add the [source directory](http://wec-sim.github.io/WEC-Sim/code_structure.html#wec-sim-source-code) (`$SOURCE`) to your path is provided in the following  [directions on how to add WEC-Sim source folder to MATLAB path](http://wec-sim.github.io/WEC-Sim/getting_started.html#step-1-add-wec-sim-source-code-to-matlab-path). An example of how this is done this is provided below. Update the path and yourUsername, then copy the following lines into the `startup.m` file (located in your `C:\Users\yourUsername\Documents\MATLAB` directory).

```
% Define the wecSimPath variable to the location of your local WEC-Sim 
% source directory as a 'string'. Copy the code in this file and place 
% it in your'startup.m' file
wecSimPath = 'C:\Users\yourUsername\Documents\GitHub\WEC-Sim\source';
addpath(genpath(wecSimPath));
```

Once the WEC-Sim source directory is added to your path, you will create a `$CASE` directory for each run, [as described here](http://wec-sim.github.io/WEC-Sim/overview.html#wec-sim-model-files). For the tutorials, `$CASE` directories have already been created for you:

- `C:\Users\yourUsername\Documents\GitHub\WEC-Sim\tutorials\RM3`
- `C:\Users\yourUsername\Documents\GitHub\WEC-Sim\tutorials\OSWEC`

For more information, refer to our [WEC-Sim Code Structure webinar and presentation](http://wec-sim.github.io/WEC-Sim/webinars.html#wec-sim-code-structure).

closed 2018-02-12 by kmruehl