WEC-Sim/WEC-Sim #231: Error when running Tutorials

genius613, 2018-02-20:
This is a follow up on the previous thread. 

I have added WEC-Sim to the Matlab path and believe I am running the tutorials from their respective folders. When I first run them, I get the following error:

[image]

Then I type wecSimInputFile = 'wecSimInputFile.m' into the Command Window, after which I get this error:

[image]

Any help here would be much appreciated.

kmruehl, 2018-02-20:
@genius613 

The first error is related to the hydrodynamic coefficients stored in the `*.h5` file. It looks like the `*.h5` file did not download properly. Please refer to the note regarding [installing git-lfs here](http://wec-sim.github.io/WEC-Sim/getting_started.html#option-3-static-code-download-not-recommended). Alternatively, you can download the `*.h5*` [files here](https://github.com/WEC-Sim/WEC-Sim/releases/tag/v3.0). 

The second error is related to running wecSim. To [execute wecSim](http://wec-sim.github.io/WEC-Sim/overview.html#step-4-execute-wec-sim) you should type `wecSim` into the command line. Running the input file does not execute the wecSim code.

genius613, 2018-02-20:
OK. I realized that the h5 files contained in the WEC-Sim zip file did not download properly. I have now downloaded them separately and the tutorials run.

Thank you so much for your help!