WEC-Sim/WEC-Sim #232: Simulation of drift forces in WEC-Sim

gui-paredes, 2018-02-21:
Hi!
Is there any way to simulate drift forces in WEC-Sim? This is very important for simulation of moored structures.
Thank you

kmruehl, 2018-02-22:
@gui-paredes Unfortunately WEC-Sim does not currently simulate drift forces. The WEC-Sim team will make note of this request and add it as a desired feature for a future release.

Tudore, 2018-11-20:
> @gui-paredes Unfortunately WEC-Sim does not currently simulate drift forces. The WEC-Sim team will make note of this request and add it as a desired feature for a future release.

Hi, how is the progress on including drift force in WEC-Sim. 
I am quite curious whether Nemoh can solve QTF or not, WAMIT requires a second order module.

gui-paredes, 2018-12-19:
Dear developers,
I will be working on including drift forces in wecsim in the next months, hopefuly in a way that can be then be permanently added to the wec-sim code. In order to do that, I would like to know if you have any suggestions for this, such as: 
create new files or modify existing ones;
which files to modify and which ones to keep intact
create wec-sim blocks or only matlab code, or combinations
any coding rules or norms that you use
and, if you are interested  in collaborating in this.

crobarcro, 2018-12-19:
Personally I would vote for pure matlab code in some form, which can then be tracked using git. Simulink models are basically binary blobs to git so version control doesn't work (ok, there are ways to make it sort of work on a machine with Matlab installed on it with special matlab versions of diff programs for simulink models, but this is awful).

In fact it would be amazing if all the hidden code in the simulink models was also kept as external functions so one could track any changes with Git.

oliviergius, 2019-02-28:
Hi,
I find your initiative of including the drift forces in WEC-Sim very intersesting and helpful. Have you managed to do something since your last post?