WEC-Sim/WEC-Sim #234: Moordyn error

Toby-D, 2018-02-26:
Has anyone else had trouble with MoorDyn lately? 

I've run simulations with MoorDyn as recently as mid January (after updating to WecSim 3.0). But now I'm getting an error from the  'InitFcn' of the MoorDyn block: 
```

    Error using loadlibrary
    Too many output arguments.

```
As of today, 2/26, I re-downloaded the Wec-Sim, MoorDyn, and Wec-Sim Applications repositories. But I still get the same error, even when I'm running the RM3 tutorial (unmodified). 

I'm running on Windows 10 (at Build 16299),
The error occurs in both R2016a and R2017b.

Any thoughts?

Thanks,

Toby. 

PS The full error text is:

```
Error using wecSim (line 250)
Error evaluating 'InitFcn' callback of SubSystem block 'RM3MoorDyn/MoorDyn/MoorDyn/MoorDyn'.
Callback string is '% Initialize MoorDyn Lib (Windows:dll or OSX:dylib)
if ismac
    loadlibrary('Lines.dylib','MoorDyn.h');
elseif ispc
    loadlibrary('Lines.dll','MoorDyn.h');
elseif isunix
  loadlibrary('Lines.so','MoorDyn.h');
else
  disp('Cannot run MoorDyn in this platform');
end
calllib('Lines','LinesInit', mooring.loc ,[0 0 0 0 0 0]);'
Caused by:
    Error using loadlibrary
    Too many output arguments.
```

Toby-D, 2018-02-27:
Update: I've looked into this some more, and the problem seems to be with the "loadlibrary" function, not specifically with MoorDyn or WEC-Sim.

kmruehl, 2018-02-27:
@Toby-D, sorry for the delayed response. This error is generated because MoorDyn is no longer included in the WEC-Sim source code. Since MoorDyn was developed independently by @mattEhall and has a separate license, we placed the [MoorDyn source code in its own repository](https://github.com/WEC-Sim/moorDyn). MoorDyn must be downloaded separately and placed in the `WEC-Sim/source/functions/moorDyn` directory to run with WEC-Sim.  Refer to the [README ](https://github.com/WEC-Sim/moorDyn/blob/master/README.md)for more info.

closed 2018-02-27 by kmruehl

Toby-D, 2018-02-28:
Thanks for the reply, Kelley.

It turned out that loadlibrary was failing (even for MATLAB's example libraries) because I had unwittingly allowed MATLAB's built-in perl.m script to shadow another script with the same name in a third-party toolbox. If anyone else ever has this problem, see:
C:\Program Files\MATLAB\R2017b\toolbox\matlab\general\perl.m

kmruehl, 2018-02-28:
@Toby-D Thanks for the heads up. Glad you were able to resolve the issue.

Tudore, 2018-11-20:
> Thanks for the reply, Kelley.
>
> It turned out that loadlibrary was failing (even for MATLAB's example libraries) because I had unwittingly allowed MATLAB's built-in perl.m script to shadow another [...]

Hi, I also encountered this problem. How can I fix the issue? Thank you!

Tudore, 2018-11-20:
> Thanks for the reply, Kelley.
>
> It turned out that loadlibrary was failing (even for MATLAB's example libraries) because I had unwittingly allowed MATLAB's built-in perl.m script to shadow another [...]

The problem comes from the conflict of the third-party toolbox WAFO. I just removed the path, and it works now.