WEC-Sim/WEC-Sim #238: simulation

stevenshine, 2018-03-19:
when i want to  run wecSim
the message box show 
[image]
what does it mean ?
how to modify?
thank u

kmruehl, 2018-03-19:
This error means that the Impulse Response Function (IRF) calculated is shorter than the convolution integral time specified in the input file. To resolve this issue you can either:
- Reduce the convolution integral time of your simulation by changing the `simu.CITime` in your `wecSimInputFile.m` 
- Rerun BEMIO and specify a longer end time (`t_end`) for the calculation of the impulse response functions. Refer to the [documentation on bemio functions](http://wec-sim.github.io/WEC-Sim/advanced_features.html#bemio-functions ) for the Radiation_IRF and Excitation_IRF.

closed 2018-03-19 by kmruehl

stevenshine, 2018-03-20:
Hi @kmruehl 
thank u help me but i still have some quesions
the message show when i run menoh
[image]