WEC-Sim/WEC-Sim #241: Diferences between Nemoh mesh and .STL for simulation on WEC-Sim

NMartinez9, 2018-03-26:
Hi everyone,

I have a similar problem descripted in #140  and #129,  simulating a bottom hinged flap. I am using Nemoh and the Mesh rutine that take account only the submerged part of the body, and the origin of coordinates at z=0. I also use the FreeCAD Salome for the geometry .STL file, but placeing the origin of coordinates at the flap's CG.
The simulation results are not as i spected so like to ask if I need to do some more changes on the Nemoh files and how I can do so.

Thank you

kmruehl, 2018-04-04:
@NMartinez9 would you be able to ZIP your case directory of the WEC-Sim run so that I can see the model? Also, if you could post a screenshot of the error your receiving, that would be very helpful. 

Thanks!

kmruehl, 2018-04-11:
@NMartinez9 it appears there are several things going on here, and I will to my best to address each of them. 

- The error you received is not due to an issue with the STL mesh, it is actually related to the `Rotational PTO Actuation Torque` block, and specifically the user defined `PTO-Sim` applied torque. When I removed the coupled `Rotational PTO Actuation Torque` and `PTO-Sim` blocks and replaced them with a simple `Rotational PTO` block and applied a linear damper (`pto(1).c `), the simulation runs as expected.  In the attached, I updated the PTO block. This means the issue is not with the WEC-Sim run, and that you'll need to look into and debug the forcing function developed in PTO-Sim to resolve the error. 
- The simulation you sent specified a convolution integral time `simu.CITime = 20;`, but you were running a `regular` wave type. In order to use the radiation convolution formulation, you must specify `regularCIC` as the wave type. In the attached, I updated the wave type. 
- In order to run WEC-Sim with non-linear hydro, you must specify `nlHydro = 1` or `nlHydro = 2`. Refer to the [documentation here](http://wec-sim.github.io/WEC-Sim/advanced_features.html#non-linear-hydrodynamics). 
- In order to run WEC-Sim with drag in pitch, you must specify `body(1).viscDrag.Drag(5,5)`. Alternatively, you can specify the drag coeff `body(1).viscDrag.cd` and characteristic area `body(1).viscDrag.characteristicArea`. 

I've update the case file to reflect these changes and included a ZIP of it here.
[Study_case.zip](https://github.com/WEC-Sim/WEC-Sim/files/1899664/Study_case.zip)

closed 2018-04-11 by kmruehl