WEC-Sim/WEC-Sim #242: Nonlinear excitation force calculation

ahmedmetin, 2018-04-01:
Hello

While I was simulating a cylindrical buoy in heave with nonlinear excitation force (nlHydro=2), I found that the final excitation force does not equate to 0 when the buoy goes completely out of the water. As far as I have figured it out the way you have implemented the total excitation force is the sum of the linear excitation force pre-computed with excitation force coefficients, nonlinear excitation force computed using the instantaneous draft of the buoy minus the linear excitation force computed with the mean draft of the buoy. I have modified the Diffraction and Excitation Function Matlab Function block to output the heave f_nonlinear and f_linear signals. You can see in the plot below that f_nonlinear gets equal to 0 when the buoy gets out of the water. (The bottom plot shows wave elevation from mean water level minus the elevation of the center of the buoy, which coincide with the CG). The total height of the buoy is 5.78m and CG = 0, i.e. when z_wave - z_buoy < -5.78/2, the buoy goes out of the water. 
Is there some inaccuracy in WEC-Sim's excitaiton force model when the buoy pops out the water or is it compensated somehow somewhere else in the model?

[image]

Thanks in advance.

yuyihsiang, 2018-04-22:
In WEC-Sim. the "total" excitation force includes  both the diffraction and the wave Froude–Krylov force. For nonlinear simulations, the basic assumption is that the diffraction part is still based on linear theory and the Froude–Krylov force is calculated based on instantaneous wave elevation and body position. Therefore, even if the body is out of the water, the diffraction part WON'T be zero. The method is a "weakly nonlinear" model, and this is just the way typically how this simple model is applied.

Yi-Hsiang

kmruehl, 2018-05-09:
@ahmedmetin does this answer your question?

closed 2018-05-10 by kmruehl