WEC-Sim/WEC-Sim #244: Center of gravity pitching flap

NicolasQu, 2018-04-24:
Hi,

I'm designing a pitching flap WEC in wec-sim, with a density equal to 440 kg/m³. 
Since the density of the body can't be changed in NEMOH, I changed its center of gravity so that bemio_NEMOH calculates the restoring coefficient correctly.
However, wec-sim visualizes the device now at the wrong position.
Now I changed the body.cg again in the wec-sim code to its correct position (and not the wrong one I used for Nemoh), but is this a correct way of working?
Normally the center of gravity is at H/2 of the device, but I used another value in Nemoh.
Or does anyone know how to change the body's density in Nemoh?

Thanks in advance!
Regards,
Nicolas

kmruehl, 2018-04-24:
For some background, when BEMIO imports hydro-data from the BEM solution (in this case NEMOH), all of the coefficients are non-dimensionalized [according to the definition here](http://wec-sim.github.io/WEC-Sim/theory.html#boundary-element-method) and saved to a `*.h5` data structure. WEC-Sim then dimensionalizes these coefficients according to the website definition and user inputs (this means you can run NEMOH with one fluid density and WEC-Sim with another). Since WEC-Sim solves all forces about each body's cg, it is very important that the cg location is accurate. Based on this fact, your proposed solution is not recommended.

Since you are unable to change the flap's mass in NEMOH directly, I recommend overwriting the flap's mass in the WEC-Sim `wecSimInputFile.m` by defining `body(#).mass` and overwriting the NEMOH calculated mass (based on displaced volume and fluid density). This is how the [OSWEC tutorial](https://github.com/WEC-Sim/WEC-Sim/tree/master/tutorials/OSWEC) is modeled. 

FYI, it's possible you can specify the flap's mass in NEMOH, but that's a question for the NEMOH developers.

NicolasQu, 2018-04-25:
Thank you for your reply!
I now tried again with fluid density in Nemoh equal to the density of the flap (being 440 kg/m³) so that the mass of the flap is correct and I set the cg location back to the correct one.
Then in wec-sim I use simu.rho = 1025. 
Will this give correct results, or is the calculation of the added mass, done by Bemio_nemoh, wrong because of calculation with a different fluid density?

NicolasQu, 2018-05-01:
Thank you for your reply!
I used the described approach and it looks like it's working

kmruehl, 2018-05-02:
@NicolasQu glad to hear that. Take care.

closed 2018-05-02 by kmruehl