WEC-Sim/WEC-Sim #245: Reference point for forces and moments

gui-paredes, 2018-04-26:
What is the reference point used by WEC-Sim to compute moments acting on a floating structure? Is it the centre of floatation, the centre of gravity, or the centre of buoyancy? Is it the same for all simulations, or can it be adjusted by the user? Different hydrodynamic codes will provide the hydrodynamic coefficients relative to different positions.

kmruehl, 2018-04-26:
@gui-paredes all forces/moments acting on bodies in WEC-Sim are calculated about their center of gravity. WEC-Sim uses [Simscape Multibody](https://www.mathworks.com/help/physmod/sm/) as its multibody solver, and thus follows its convention for all forces/moments to be applied at the cg. On a related note, this means that the BEM solution used by WEC-Sim must be solved for the body with the origin at its cg.

kmruehl, 2018-05-09:
@gui-paredes does this answer your question?

gui-paredes, 2018-05-11:
It does, but I I have a new one. The mooring forces are applied at the water level + the initial linear displacement of the bodies. Are the moments of these forces then computed at the centre of gravity?

kmruehl, 2018-05-11:
@gui-paredes, in WEC-Sim all forces and moments are calculated and applied at the cg for each body. If a force/moment is physically applied at a location other than the cg (i.e. mooring/joints), we use Simscape `Rigid Transform Blocks` to account for this offset from the cg, and transform the force/moment to be applied at the cg. As a result, all forces in the WEC-Sim `output` data structure are based on the body's cg. I hope this helps clarify.

closed 2018-05-30 by kmruehl