WEC-Sim/WEC-Sim #246: Can we use WEC-SIM for the projects like Swanseabay Tidal lagoon?

bobbyj2492, 2018-04-30:
Hello everyone!!

Firstly, I would like to thank everyone for developing a code suits for analysing marine renewable energy models. My question here is, can we use WEC- SIM for modelling and analyzing projects like [Swanseabay Tidal lagoon ](https://www.youtube.com/watch?v=nne4sSgRfTQ) and  [Larance tidal power stations](https://www.youtube.com/watch?v=ZgwspUPLafQ) and  [Dynamic Tidal Power](https://www.youtube.com/watch?v=W7PTxoW66dM&t=414s). The common one in all of them is a fixed structure called dam which houses the turbines and allows the water flow when gates or valves open similar to Hydro electric power. Really appreciate your response.

Thank you
Ravindra

kmruehl, 2018-05-10:
@bobbyj2492 WEC-Sim is a code developed to model ocean wave energy converters, not for tidal turbines. I do not think there are a good application cases for WEC-Sim. Thank you for your interested.

closed 2018-05-10 by kmruehl

bobbyj2492, 2018-05-11:
@kmruehl  Thank you so much for your response. Even I noticed the same after integrating WEC SIM in MATLAB. By chance, do you have any idea of other tools to model tidal lagoons or barrages? Thank you.

kmruehl, 2018-05-11:
@bobbyj2492 this is a good question, but unfortunately I am unaware of such codes (although I'm sure they exist).