WEC-Sim/WEC-Sim #249: Wec-sim .h5 file reading error

kennybear1111, 2018-05-08:
Hello,

I tried to run the RM3 model from the tutorial provided by Wec-sim on MATLAB 2016b. After opening startup.m and wecsiminputFile.m, I typed "wecSim" in the command window and get the error as shown below:

WEC-Sim Read Input File ...   
Error using h5readc
Unable to open file. Filename may be corrupt or have unsupported characters.

Error in h5read (line 58)
[data,var_class] = h5readc(Filename,Dataset,start,count,stride);

Error in bodyClass/readH5File (line 84)
            obj.cg = h5read(filename,[name '/properties/cg']);

Error in wecSim (line 85)
        body(ii).readH5File;

I also tried two other PCs with MATLAB 2016b and got same error. My friend's laptop with MATLAB 2017b did not have the error. We did the same procedure to run the simulation. May I ask what will be the reason for this error? Is there any way to solve the issue with MATLAB 2016b?

Thank you
Qiuchi

talhame07, 2018-05-10:
Hi,
I am using Matlab 2017a. Can I use WEC-sim in my Matlab version?

kmruehl, 2018-05-10:
@kennybear1111, yes you can run WEC-Sim with MATLAB v2017a. The issue is that unless you have git-lfs installed, you will not have the proper `*.h5` files. You can either install [git-lfs](https://git-lfs.github.com/), or download the `*.h5` files directly from the release page [here](https://github.com/WEC-Sim/WEC-Sim/releases/tag/v3.0). We use git-lfs because the files are very large (~32MB and 7MB), and want to keep the repository size down.

talhame07, 2018-05-14:
Hi Kelley Ruehl,
Can I run WEC-Sim with 2018a?

kmruehl, 2018-06-04:
@kennybear1111 that is correct. If git-lfs is not installed, the downloaded `*.h5` file will only be ~1kb. Downloading the files manually as outlined in Option 3 is one way of resolving the issue. The other resolution is to install [git-lfs](https://git-lfs.github.com/) and use Option 1. Thank you for the suggestion, we [added an error flag in the wec-sim source code](https://github.com/WEC-Sim/WEC-Sim/commit/101139582603abee9b487393956669faa9d2ac17) and will update the documentation.

closed 2018-06-04 by kmruehl

abhisheks865, 2020-11-17:
Hi there,
When I am running the wecSim code after creating h5 file from AQWA, I am getting below error. 

[image]

I have checked my H5 file and it is of 8 MB so i think it has all the data required to proceed in wecSim. Can someone help me to resolve this error?