WEC-Sim/WEC-Sim #250: Error while running WecSim.

Krishnendu-Ocean, 2018-05-15:
Hello,

I am trying to model a spherical model (only the submerged part is represented) in WEC-Sim. The hydrodynamic coefficients are modelled in Nemoh-open source software and using BEMIO, the .HDF5 file is successfully created. I am getting following errors when I run wecSim;
Case 1: The time period provided in Nemoh model is 1s to 2.3s
WEC-Sim Pre-processing...
Error using wecSim (line 143)
simu.CITime is larger than the length of the IRF
Case 2: The time period in Nemoh was changed to 3s to 20s
Error using wecSim (line 136)
waves.w outside of range of available hydro data

I tried changing the simu.endtime, simu.dt and simu.rampT in the wecSim input file. Someone please help me to solve this issue. Please note that I am trying to model the lab-scale parameters with wec-sim. 
Thank you.
Krishnendu

nathanmtom, 2018-05-30:
Krishnendu,
Thank you for your comments. Based on the information provided we expect your issues are due to the following:

Case 1: When you ran Bemio the IRF length for the radiation and excitation force kernels is set at 30 s in the default settings. In the WEC-Sim, the default simu.CITime is 60 s unless otherwise defined in the wecSimInputFile.m. Please try to add simu.CITime = 30 in your wecSimInputFile.m or to the length you defined in BEMIO.

Case 2: It would appear that the wave period you selected, waves.T, is outside the range of hydrodynamic values calculated in NEMOH. If the wave period is below 3 s or above 20 s, WEC-Sim will not extrapolate to obtain the hydrodynamic coefficients and the simulation will provide an error; however, if it is between 3s-20s then it will interpolate between the data points provided by WEC-Sim. If possible provide more information on the environmental conditions you are simulating.

If the comments above do not solve your questions, please provide greater detail on the simulations you are running, even potentially uploading your wecSimInputFile.m.

Cheers,
Nathan Tom
National Renewable Energy Laboratory

closed 2018-06-08 by kmruehl

Krishnendu-Ocean, 2018-06-26:
Dear Sir,
Thank you for your comments. Error was sorted by adding simu.CITime = 20 in the wecSimInput file. I could proceed with the regular wave simulations.

When I try to do a free decay test (noWaveCIC) on the model, I am getting following error;
“Reference to non-existent field 'simulation_parameters'.
Error in wecSim (line 105)
waves.waveSetup(body(1).hydroData.simulation_parameters.w,
body(1).hydroData.simulation_parameters.water_depth, simu.rampT, simu.dt, simu.maxIt, simu.g,
simu.endTime);”

I am using Nemoh for BEM calculations, Is it something to do with the .h5 file? I am using the same .h5 file as used for regular wave testing. I have followed the IEA OES Task10 free Decay example file for creating the input file. So I hope it is not the error with the wecSimInput file. Thank you in advance.

Krishnendu-Ocean, 2018-07-05:
Thank you for your help. The error in the simulation is sorted. However, the free decay profile is not being generated as required. I have attached the plots for response of the body, the Simulink model, the wecSimInput file and the .h5 file herewith.
I have one more query. If I have to model a PTO system placed above the model (no contact with water) what should be done in the Simulink model. Is it possible to simulate any such WEC model? 

[DecayTest.zip](https://github.com/WEC-Sim/WEC-Sim/files/2165192/DecayTest.zip)

nathanmtom, 2018-07-12:
Dear User,
Apologies in the delay in our response. Unfortunately, each time I try to download the .zip, the file is empty so we cannot review any of the documents. Perhaps, double check your .zip file or upload each file separately.

It is possible to model a PTO system placed above the water surface. You will need to define the pto.loc to be above the water line (positive z value) and then calculate the orientation by defining pto.orientation.z to have it point down towards the body or in what orientation you desire.

More information can be found here: https://wec-sim.github.io/WEC-Sim/advanced_features.html#constraint-and-pto-features

Thank you,
Nathan Tom
National Renewable Energy Laboratory

Krishnendu-Ocean, 2018-07-14:
Dear Sir,
I am attaching the .zip file again.  Please find the same.
I could change the orientation of pto. and work with the model. Thank you.
[DecayTest.zip](https://github.com/WEC-Sim/WEC-Sim/files/2194759/DecayTest.zip)

Regards,
Krishnendu

nathanmtom, 2018-07-18:
Dear User,
Thank you for the updated .zip file. After reviewing your files we have identified several issues:

1. The parameter, simu.CITime = 10, was your initial value, but it would need to be increased to 30 s or as long as it takes for the impulse response function to decay to 0.
2. If your simulink model, the constraint had the base and follower directions reversed, the correct orientation should be as shown in the attached image.
3. The error appears to be in the construction of your impulse response function, the CIC calculation that is causing the motion to not decay as expected from the noWaveCIC case. By selecting the noWave case, the heave response follows the expected motion of the decay test. The difference between the two simulation cases is how the radiation forces are calculated. We suggest revisiting your NEMOH solution and refine the mesh and double check your inputs are correct. It is preferable to calculate the hydrodynamic coefficients to a high enough frequency such that the radiation wave damping goes to zero. In your coefficients the values are no zero at the highest wave frequency.

[image]
[image]

Please let us know if the above suggestions help in solving your simulation problems.

Cheers,
Nathan Tom
National Renewable Energy Laboratory

Krishnendu-Ocean, 2018-07-20:
Dear Sir,
Thank you. 
As suggested I will revisit the Nemoh file and try running the nowaveCIC condition.
Regards,
Krishnendu