WEC-Sim/WEC-Sim #252: etaImport wave simulation

Ale4c, 2018-05-31:
Hello,

I have a couple of question about using etaImport to define wave elevations.
With this function you can define the time-series of wave elevations in the origin of the coordinate system.

What I wonder is, what does Wec-Sim exactly do with this values? Does it interpolate the wave elevation through the other points of the domain, creating a sort of waves surface? Does it just consider one wave into the origin? I can see the WEC's bodies moving, but I'm not really understanding what kind of wave is moving each of them in each time-step.

Also, are the waves moving from the origin towards any direction?

I could not find the part of code in Wec-Sim that does all of this.

I also tried to simulate the imported waves using Paraview, but it does not seem to work. Does Paraview not support etaImport? The 'waves' files generated into vtk folder contain just zero values.

I tried also to measure the wave elevation in other points of the domain using virtual wave gauges, but again the output is just all zeroes. Is etaImport not supported again?

Thanks for your help.

Ale4c, 2018-06-05:
"The reason for this is that wave surface elevation is not necessary to calculate the forces acting on each body. Since the BEM solution is solved for each body located relative to the global coordinate system with the origin at the SWL, all of the wave forces are based on the wave surface elevation at the origin. "

Thanks a lot for your answer. 

If all I am defining is the wave elevation in the origin, does WEC-Sim not take in account wave shape/length for the calculation of the forces acting on each body (assuming multiple bodies along the x-axis)?   

Thanks again

kmruehl, 2018-06-05:
@Ale4c The way WEC-Sim handles the wave forces on bodies offset from the origin (i.e. along the x-axis) is through the Boundary Element Method (BEM) solution. The BEM solution is imported into WEC-Sim and used to calculate wave loads on each body (i.e radiation, excitation). It is actually the BEM solution that accounts for the wave loading on each body relative to the origin at the Still Water Line (SWL). For example, if a body is offset from the origin along the x-axis, the wave excitation from the BEM solution has a magnitude and phase. The phase accounts for the the body's offset from the origin (i.e. accounting for the time it takes for the wave to propagate to the body along the x-axis). Since the BEM solution solves for all wave loads relative to an origin at the SWL, WEC-Sim only requires definition of the wave surface elevation at the origin. I hope this helps.

kmruehl, 2018-06-08:
@Ale4c does this answer your question?

closed 2018-06-08 by kmruehl

Ale4c, 2018-06-08:
Yes, thanks a lot for your prompt help