WEC-Sim/WEC-Sim #253: BEMIO - Read_AQWA - wrong density

HathiHathi, 2018-06-06:
Converting my AQWA files in BEMIO, i got a false output that was caused by a wrong density value.

Recent modification in the Read_AQWA.m checks for version 18.2. And at line 71 (Read_AQWA.m) it used the version to select the location of the density in the GENERAL block. There i found out that a different line of the raw1 data was used for the density coefficient.

In my opinion this should be solved by adding the next command to line 72 and 77:
`            tmp = str2num(raw1{ln+1});`

(Or  adding this command one time at line 70, probably also works)

As a result, the code, that worked for me, becomes:

```
for ln = n:length(raw1);
    if isempty(strfind(raw1{ln},'GENERAL'))==0
        if V182 == 1  % General information columns in Versions>18.2
            tmp = str2num(raw1{ln+1});
            hydro(F).h      = tmp(2);   % Water depth
            hydro(F).rho    = tmp(3); % Water density
            hydro(F).g      = tmp(4);   % Gravity
        else          % General information columns in Versions<18.2
            tmp = str2num(raw1{ln+1});
            hydro(F).h      = tmp(1);   % Water depth
            hydro(F).rho    = tmp(2); % Water density
            hydro(F).g      = tmp(3);   % Gravity
        end
        clear V182
    end

```

HathiHathi, 2018-06-06:
Another issue with Read_AQWA.m in the code at line 91:

The draft of the object is not saved for use. And I think it should be saved, by including it to `hydro(F).draft`, because otherwise the object floats in a random position in space. Draft is not mentioned elsewhere in the original code, so no other function is using it right now.
 
In my opinion, the original code should be changed, at line 90-95, to:
```
    if isempty(strfind(raw1{ln},'DRAFT'))==0
        for i=1:hydro(F).Nb
            tmp = str2num(raw1{ln+i});
            hydro(F).draft(i) = tmp(2); % Draft
        end
    end
```

ThommyBee, 2018-06-13:
Hi HathiHathi,

Thanks for pointing that out. At one point these changes were made but it must have been dropped somewhere along the way.

You are right with adding `tmp = str2num(raw1{ln+1});` into the section where Read_AQWA.m reads waterdepth, water density, and g. Same applies to the other sections 'Draft' and 'COG' but here the code snipped to skip an additional line is from what I can see included in the current Read_Aqwa.m file.

As far as I can see, when using V18.2 and above there are additional sections in the Aqwa AH1 file that do not occur in earlier versions. The sections are labeled 
ADDEDMASS_LF, DAMPING_LF, and ADDEDMASS_HF for additional high and low frequency added mass, respectively added damping.

These sections are being read as regular "added damping" and "added mass" sections in the current code which can lead to matrix sizing errors when running the script:

Currently:
```
 if (isempty(strfind(raw1{ln},'ADDEDMASS'))==0 | isempty(strfind(raw1{ln},'DAMPING'))==0)
        if isempty(strfind(raw1{ln},'ADDEDMASS'))==0 f = 0; else f = 1; end
        for m=1:hydro(F).Nb
            for k=1:hydro(F).Nb
                for j=1:hydro(F).Nf
                    for i=1:6
                        tmp = str2num(raw1{ln+(m-1)*hydro(F).Nb*hydro(F).Nf*6+(k-1)*hydro(F).Nf*6+(j-1)*6+i});
                        if length(tmp)>6 ind = tmp(1:3); tmp(1:3)=[]; end
                        if f==0 % Added Mass
                            hydro(F).A((ind(2)-1)*6+i,((ind(1)-1)*6+1):(ind(1)*6),ind(3)) = tmp;
                        elseif f==1 % Radiation Damping
                            hydro(F).B((ind(2)-1)*6+i,((ind(1)-1)*6+1):(ind(1)*6),ind(3)) = tmp;
                        end
                    end
                end
            end
        end
```

What helps is including another case distinction between the HF/LF frequencies and the regular AM/AD sections:

```
if ((isempty(strfind(raw1{ln},'ADDEDMASS'))==0 && isempty(strfind(raw1{ln},'LF'))==1 && isempty(strfind(raw1{ln},'HF'))==1) |...
            (isempty(strfind(raw1{ln},'DAMPING'))==0 && isempty(strfind(raw1{ln},'LF'))==1))
        if isempty(strfind(raw1{ln},'ADDEDMASS'))==0 f = 0; else f = 1; end
        for m=1:hydro(F).Nb
            for k=1:hydro(F).Nb
                for j=1:hydro(F).Nf
                    for i=1:6
                        tmp = str2num(raw1{ln+(m-1)*hydro(F).Nb*hydro(F).Nf*6+(k-1)*hydro(F).Nf*6+(j-1)*6+i});
                        if length(tmp)>6 ind = tmp(1:3); tmp(1:3)=[]; end
                        if f==0 % Added Mass
                            hydro(F).A((ind(2)-1)*6+i,((ind(1)-1)*6+1):(ind(1)*6),ind(3)) = tmp;
                        elseif f==1 % Radiation Damping
                            hydro(F).B((ind(2)-1)*6+i,((ind(1)-1)*6+1):(ind(1)*6),ind(3)) = tmp;
                        end
                    end
                end
            end
        end
    end
```

Another solution to this is to uncheck the option to calculate HF and LF frequencies in Aqwa itself. You can find it under Analysis Settings - Common Analysis Options.

Hope this helps
Thomas

nathanmtom, 2018-06-14:
@HathiHathi @ThommyBee Thank you for your comments and analysis in response to the BEMIO Read_AQWA.m. Upon review of the issue and suggested changes, the WEC-Sim team agrees that code modifications need to be made to import the correct water density and account for the new sections on describing added mass and wave damping. We have gone ahead and incorporated the suggested code changes and have pushed them to the master branch.

@HathiHathi In response to your question about saving the draft in the hydro object, we do not save the draft as it is not used elsewhere in WEC-Sim. The position in the visualization is defined by the center of gravity of the body. Please check to ensure you have defined the correct center of gravity.

If you have additional comments or questions please let notify the development team.

Cheers,
Nathan Tom
National Renewable Energy Laboratory

closed 2018-07-18 by kmruehl