WEC-Sim/WEC-Sim #255: Initialize simulation class

TheGreatBumpkin, 2018-06-11:
What does it mean to initialize the simulation class? When I try running wecsim.m or wecsiminputfile.m I get an error message saying that the function "simulationClass" is undefined. I don't know if I am supposed to put something between the parentheses.

nathanmtom, 2018-06-14:
Thanks for your question.  Here are some action items we suggest based on the information you provided.

1) [Add WEC-Sim Source Code to MATLAB Path](https://wec-sim.github.io/WEC-Sim/getting_started.html#step-1-add-wec-sim-source-code-to-matlab-path). Then restart MATLAB.

2) Are you currently attempting to run a tutorial example? If so, can you let us know which one?
     - All tutorial files should have the correct syntax to initialize the simulation class in the wecSimInputFile.m
     - If you are constructing your own wecSimInputFile.m, then make sure to include the following before defining any of the simu variables: "simu = simulationClass();"

If these two actions do not solve your issue, please provide more information on the type of simulation you are running.

Cheers,
Nathan Tom
National Renewable Energy Laboratory

TheGreatBumpkin, 2018-06-16:
I was running the RM3 tutorial simulation. I tried running simulationclass.m and that solved the problem. I realized that I had not properly added the Source file to the MATLAB path. I corrected this and tried running it again. I no longer have the problem with the "simu = simulationClass()" being undefined. 
However, I am now receiving the following error messages:
Error using h5readc
The filename specified was either not found on the MATLAB path or it contains unsupported characters.

Error in h5read (line 58)
[data,var_class] = h5readc(Filename,Dataset,start,count,stride);

Error in bodyClass/readH5File (line 84)
            obj.cg = h5read(filename,[name '/properties/cg']);

Error in wecSim (line 94)
        body(ii).readH5File;

nathanmtom, 2018-06-19:
Thank you for your response. Glad to hear you were able to resolve the issue with the simulation class.

I suggest reprocessing the .h5 file by rerunning BEMIO on the hydrodynamic data. It is possible the .h5 wasn't downloaded properly. 

If that does not solve your problem, please let us know.

Cheers,
Nathan Tom
National Renewable Energy Laboratory

TheGreatBumpkin, 2018-06-22:
Thank you. that solved the hydrodata problem. One last problem, I am receiving the following message.

Error using wecSim (line 259)
Error evaluating parameter 'TranslationCartesianOffset' in 'rm3/Global Reference Frame/Rigid
Transform'
Caused by:
    Error using wecSim (line 259)
    Dot indexing is not supported for variables of this type.

nathanmtom, 2018-07-02:
Dear User,
Glad to hear the previous correction solved your previous issue.

A few questions:
1. You are still running the RM3 tutorial simulation?
2. What version of Matlab are you running?
3. Are you running a PC or MAC operating system?

It would appear the issue arises from how the constraints are called and assigned. With the above information we can try to see if there has possibly been a change in code that needs to be adjusted.

Also, if you feel like it you could .zip your folder with the .slx file, the wecSimInputFile, and the .h5 and send it to use to see if we can reproduce the error.

Cheers,
Nathan Tom
National Renewable Energy Laboratory.

TheGreatBumpkin, 2018-07-04:
I am still running the RM3 tutorial. I am running Matlab 9.4, Simulink 9.1,
Simscape 4.4, and Simscape Multibody 5.2. I am running on a PC

TheGreatBumpkin, 2018-07-10:
[RM3.zip](https://github.com/WEC-Sim/WEC-Sim/files/2179703/RM3.zip)

nathanmtom, 2018-07-12:
Dear User,
Thank you for the .zip file. We traced the issue down to the Simulink Model File. In your file you were attempting to use the .slx version, while a conversion was made in March to move to the .mdl format. If you delete the .slx file, keep the RM3.mdl and uncomment " simu.simMechanicsFile = 'RM3.mdl';" the RM3 tutorial will run as designed.

Please let us know if you are still having problems after the above fix.

Cheers,
Nathan Tom
National Renewable Energy Laboratory

kmruehl, 2018-07-23:
@TheGreatBumpkin did this resolve your issue?

TheGreatBumpkin, 2018-07-23:
yes

Legaze, 2018-11-29:
Hello, 
I am having the same error problem:

Simulating the WEC device defined in the SimMechanics model RM3.slx...   
Error using simulationClass/loadSimMechModel (line 87)
block_diagram does not have a parameter named 'SimMechanicsOpenEditorOnUpdate'

Error in wecSim (line 256)
simu.loadSimMechModel(simu.simMechanicsFile);

I have tried what it is mention but the problem still continues, I am using the latest version of Wec-Sim and Matlab 2018.

Do you have any other solution?

Legaze, 2018-11-29:
Also to add, I keep having the same problem in the OSWEC tutorial.
Could you help me?

kmruehl, 2018-11-29:
@Legaze i just pulled the latest version of WEC-Sim from master and ran the OSWEC and RM3 tutorials on my laptop without issue, however I have 2017b. I have not run WEC-Sim with 2018, but it should work just the same. 

Have you pulled the latest version of WEC-Sim from master? https://github.com/WEC-Sim/WEC-Sim commit  b3a1975?

Legaze, 2018-11-30:
Yes :(, I have the newest one as well and still I have the same problem.
I think it should work but it does not, I have tried already both adn same error, I also check the simulation class funstion but everything seems to be normal.
I do not understand why it keeps telling me that the block diagram is missing the 'SimMechanicsOpenEditorOnUpdate' parameter. :/

[image]

kmruehl, 2018-11-30:
Have you verified that you have all of the required MATALB toolboxes?

And have you added the `WEC-Sim/source` directory to your MATLAB path?

kmruehl, 2018-12-13:
@Legaze it looks like this error is because you need to clone the MoorDyn repository into its WEC-Sim source directory. MoorDyn is hosted on a separate MoorDyn repository. It must be download separately, and all files and folders should be placed in the $Source/functions/moorDyn directory.

For instructions on how to do this refer to our website:
http://wec-sim.github.io/WEC-Sim/advanced_features.html#mooring-features

closed 2018-12-13 by kmruehl

sricharanvvs, 2019-06-07:
Hi @nathanmtom I am referring to the starting part of the thread.
I have error: 
Error using simulationClass/loadSimMechModel (line 83)
block_diagram does not have a parameter named 'SimMechanicsOpenEditorOnUpdate'

Error in wecSim (line 210)
simu.loadSimMechModel(simu.simMechanicsFile);
 
The simulink file is customized one not the standard. I am using 2018b version. As you suggested I tried to change the extension to .mdl from .slx . This didnt work.
But the whole thing worked very well with Matlab 2015b. I am getting other errors in 2017b.
Can you please help me through this? (In all the
 cases the windows is the operating system)
File is 40MB unable to upload here as .zip etc even after compression

sricharanvvs, 2019-06-07:
The error in case of MATLAB2017b
Simulating the WEC device defined in the SimMechanics model h_and_f_rev2.slx...   
Error using error_for_persistent_vars_in_ct_mlfcn (line 9)
Instantaneous FS uses constructs that are invalid when the block specifies or inherits a continuous sample time. Invalid
constructs include the use of persistent or global variables, calls to exported functions or using coder.extrinstic.

Error in sfprivate (line 15)
       feval(fcnName,varargin{:});

Error in wecSim (line 211)
sim(simu.simMechanicsFile);
Suggested Actions:
    • Set sample rate of this block to inherit the fastest discrete rate in the model. - Fix
    • Open the properties dialog of this block to set the sample time manually. - Open
 
Cannot change property while iced

Error in set_sample_time_to_fastest_discrete_rate (line 7)
    chartH.ChartUpdate = 'DISCRETE';

Error in sfprivate (line 15)
       feval(fcnName,varargin{:});

Aiswariak, 2022-05-23:
I have been using WEC-Sim 4.2 version and has now updated to 4.4 version. My MATLAB is R2021b. When I am trying to run wecsiminputfile.m, I get an error message as follows:

Unrecognized function or variable 'simFile'.

Error in wecSim (line 46)
sim(simFile, [], simset('SrcWorkspace','parent'));

The same code works fine with the older version. I could also run RM3 tutorial successfully. Kindly help.
[image]