WEC-Sim/WEC-Sim #256: h5 File Reading

sevenseasjim, 2018-06-18:
Hello,
I'm working my way through the tutorials. In the RM3 folder, the rm3.h5 file is not readable in a HDH5 viewer. Matlab will throw an error at the h5readc step. I reprocessed the h5 file using the bemio.m file. Then wecSim would work and the file was viewable in HDH5 viewer.  I am now seeing the same issue with the h5 files in the RM3_Mooring example in the Wec-Sim Applications Master folder. Has the h5 format changed since you uploaded these examples?
Thanks
Jim

jmdearman1, 2018-06-20:
I had the same issue and solved it by installing git-lfs and creating a new local clone. It is mentioned in the docs but could probably be made more clear that it is required to use the tutorial h5 files.

`git lfs install`

kmruehl, 2018-07-18:
@sevenseasjim thank you for your question. The issue is due to the fact that we use git large file storage (git lfs) to host the `*.h5` files instead of hosting them directly on the WEC-Sim GitHub repository. The solution @jmdearman1 suggested is one way of resolving this issue - to install git lfs and thus  access the correct `*.h5``files. Your solution of rerunning the `bemio.m` script to generate a new `*.h5` is an equally good way of resolving the issue. 

The RM3_Mooring application case has the same issue. The `rm3.h5` file for this case is hosted on git lfs to save space (as shown in the image below). As such, I recommend installing git lfs to access the correct `*.h5` file. If you already have git lfs included in your version of Git, this is done by typing `git lfs install` into your Git BASH. However, if you do not already have git lfs in your version of Git, you will need to download it from https://git-lfs.github.com/ first, and then install it.

[image]

kmruehl, 2018-07-18:
Also worth noting, I just updated the wec-sim documentation and added an error flag to the WEC-Sim executable providing this same guidance (to install git lfs or to run bemio.m).

kmruehl, 2018-07-26:
@sevenseasjim did this resolve your issue?

sevenseasjim, 2018-08-02:
Sorry for delay but I hadn't heard any response so I moved on. I am still having trouble in Windows getting this to work.

kmruehl, 2018-08-02:
@sevenseasjim did you try rerunning the bemio script?

sevenseasjim, 2018-08-03:
Hi, I figured out what was going on with the git clone lfs process. Thanks for your help.

closed 2018-08-03 by sevenseasjim

kmruehl, 2018-08-03:
@sevenseasjim glad you were able to resolve the issue!