WEC-Sim/WEC-Sim #258: Change in linear hydrostatic restoring force

annetbr, 2018-07-16:
Hi, 

I recently updated from WEC-Sim version 2.1 to 3.0 and noticed that there is a change in the hydrodynamic body block. I am using the linear hydrostatic restoring force. The figures below show the models as they are given in v2.1 and v3.0 respectively. As I understand the change, a moment is added because the buoyancy force and gravity force may be applied in different locations (in the x-y-plane). Could you explain why this moment was added and in short the theory behind? Mainly, I am curious about two topics: 

1. How does this relate to the forces calculated from the restoring matrix? Is it an additional moment, that is not taken into account when calculating the restoring coefficient matrix (for example due to some some simplifications)? 
2. The moment seems to be constant. Will it not change with the angles of the body (roll, pitch and yaw)? 

Thank you very much in advance! 
Best regards, 
Annette

[image]

[image]

nathanmtom, 2018-07-18:
Annette,
Thank you for your question. You are correct that a modification was made to add an additional moment to the body, if the center of gravity (COG) and center of buoyancy (COB) are not aligned in the x-y plane. If the COG and COB are not aligned, then a moment is generated about the COG because of the buoyancy forces that will want to force the body to rotate until the lever arm goes to zero. In this reorientation the COG and COB will now align in the x-y plane. In response to your specific questions:

1. This moment does not influence the calculation of the restoring matrix as calculated from WAMIT, NEMOH, AQWA or other BEM solution. It is a moment rather than moment per rotation that would be placed in the restoring coefficient matrix.
2. The moment as calculated will be constant during the simulation regardless of rotation of the body. You are correct that the moment should be updated based on the rotation of the body; however, as noted in the response above, if you have a floating system the body will continue to rotate until the COG and COB are aligned. This may cause the submerged portion of the body to change significantly as well and shift the COB. We decided to keep it constant and base the calculation about the equilibrium position to be consistent with the other assumptions that come from linear wave theory.

I hope this answers your questions, but please let us know if you have other questions or concerns.

Cheers,
Nathan Tom
National Renewable Energy Laboratory

annetbr, 2018-07-19:
Hi Nathan, 

Thank you for your reply. Yes, this pretty much answers my question. 

As I understand it, this moment was added for the case where the wave energy converter has a COB offset relative to the COG (in x-y-plane), with the assumption that there are other, non-hydrodynamic, moments in the model counteracting it. The sum of all these moments will give the device the desired equilibrium position, which is not the position that aligns COG and COB vertically. It is assumed that the simplifications of linear wave theory apply. Small angles (from this equilibrium position) can be assumed and the moment will be approximately constant. Is this correct or am I misunderstanding something?

Thanks again, 
Annette

nathanmtom, 2018-07-19:
Annette,
Glad our response answered your question. Yes, a counteracting moment is necessary to keep the device at the desired equilibrium. The small angle approximation is also assumed such that the moment that results form the offset in COB and COG (in x-y-plane) is modeled as constant in the simulation. Please let us know if you have other questions or need additional clarification.

Cheers,
Nathan Tom
National Renewable Energy Laboratory

annetbr, 2018-07-20:
Hi Nathan Tom,

That answers my question. Thank you very much!

Best regards, 
Annette

closed 2018-07-23 by kmruehl