WEC-Sim/WEC-Sim #261: Imported wave elevation time history

ttosdevin, 2018-08-17:
Hi, I'm having some issues with the eta import waveclass. I am modelling a single buoy in heave. While there is no error flagged and the wave elevation output matches the input, the buoy position output is far smaller than it should be (close to zero). 

The RM3 tutorial runs fine and so do the other wave classes. I ran a regular wave under the regular wave class and the buoy positions were as expected. I then saved the wave elevation and time outputs and used them as the etaimport input and the same thing happened, the buoy position is around 10x less than when I ran the same wave for the regular wave class. 

Any help is much appreciated,
Thanks

yuyihsiang, 2018-09-04:
Hi 

I have just did the exact same thing as you did and was not able to re-produce your issue. The attachment is the case I have tried. 

[RM3.zip](https://github.com/WEC-Sim/WEC-Sim/files/2349069/RM3.zip)

Let me know if you still have issue on this.

Yi-Hsiang

ttosdevin, 2018-09-06:
Hi,

Thanks for the response, sorry my description of the problem wasn't very
clear.
The problem is occurring not for the tutorials, but for my inputs (see
attached the irregular wave run details). The regular wave class inputs run
fine but if I input the same wave using eta import the responses are very
different (I'm trying to run irregular waves using eta import I just did
this to try to determine what the problem is).

The stl mesh I am using doesn't import to WECSim properly and appears far
bigger than it should, it looks like it could be 25m in diameter rather
than 25mm (this is probably an issue with my mesh rather than WECSim). This
doesn't really matter to me though as it shouldn't affect the WECSim output
as the stl mesh isnt used in the calculations for the linear model? Or does
the eta import require an accurate stl mesh?

Thanks,
Tom

kmruehl, 2018-09-06:
@ttosdevin can you attach the WEC-Sim model run so that we can reproduce the error? Thank you.

You are correct, unless you are running WEC-Sim non-linear hydro, the STL mesh is not used by the code other than for visualization.

ttosdevin, 2018-09-06:
[IW 2.zip](https://github.com/WEC-Sim/WEC-Sim/files/2358214/IW.2.zip)
Ah sorry here it is.

nathanmtom, 2018-09-07:
@ttosdevin After reviewing the files you have submitted, I believe that somehow the scaling of the wave-excitation impulse response function is incorrect. Therefore, the issues appears to reside in using BEMIO. When running the userDefined option your output excitation values will be much smaller because of the scale of the impulse response function that must be convoluted with the incident wave elevation. In order to further explore this issue, please provide the following information:

1. What software was used in calculating the hydrodynamic coefficients?
2. What options were used when calculating the impulse response functions in BEMIO?  If you have the folder and all of your inputs that would be helpful in review.

If you have questions or need clarification please let us know.

Cheers,
Nathan Tom
National Renewable Energy Laboratory

ttosdevin, 2018-09-08:
Ah thank you! It is working now, I had not set the frequencies correctly in
BEMIO.

kmruehl, 2018-09-10:
@ttosdevin I'm glad that this resolved your issues. @nathanmtom thanks for identifying the problem.

closed 2018-09-10 by kmruehl