WEC-Sim/WEC-Sim #262: WEC-SIM has something wrong about Loading the AQWA ouput file

Zhao900429, 2018-08-17:
I have some troubles about  Loading the AQWA output file. I don't know why WEC sim can not read it. I use the  BEMIO.m function applied by the example but it do not work.
**clc; clear all; close all;
hydro = struct();
hydro = Read_AQWA(hydro,'C:\Users\39014\Desktop\test\5.AH1','C:\Users\39014\Desktop\test\5.LIS');
hydro = Radiation_IRF(hydro,15,[],[],[],[]);
hydro = Radiation_IRF_SS(hydro,[],[]);
hydro = Excitation_IRF(hydro,15,[],[],[],[]);
Write_H5(hydro)
Plot_BEMIO(hydro)**

The MATLAB will report errors like these:

**Subscripted assignment dimension mismatch.

Error in Read_AQWA (line 182)
                hydro(F).sc_ph((kdiff-1)*6+1:kdiff*6,j,i) = tmp2;  % phase of scattering force**

Could anyone please help me?
My email is chenyu.zhao@plymouth.ac.uk
thanks a lot !!!

HathiHathi, 2018-08-28:
Hi,

My first check would be if your multiplication signs are there, cause in your comment it looks like the * is lost. But most likely that is a result of copy-paste to this post. 
So just checking, is it:
`hydro(F).sc_ma((kdiff-1)*6+1:kdiff*6,j,i) = tmp1;`

Then you could check the location of input for kdiff at line171:
`            kdiff=str2num(raw2{ln-2}(end-9:end-8));  `

Or otherwise check the input for tmp1:
` tmp1 = str2num(raw2{ln+6+(i-1)})`

kmruehl, 2018-09-04:
@Zhao900429, what version of AQWA are you using? And have you pulled the latest version of WEC-Sim? This issue could be caused by to the updates made to WEC-Sim to read the latest AQWA version's data file. The `BEMIO.m` script was updated because the AQWA output structure has changed across versions, and there may be an AQWA version compatibility issue.  Also, Would you be willing to share your AQWA output files so that we can reproduce your error? Thank you. 

@HathiHathi thanks for your help!

Zhao900429, 2018-09-04:
Hi,
Thank you very much for your help, I have solved the problems. I think something wrong with the code in WEC-sim 3.0 to read the AQWA output file. I replaced it by the function in the WEC-sim 2.0 and it does work!  I will check the code in 3.0.

All the best,
Chenyu

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

This email and any files with it are confidential and intended solely for the use of the recipient to whom it is addressed. If you are not the intended recipient then copying, distribution or other use of the information contained is strictly prohibited and you should not rely on it. If you have received this email in error please let the sender know immediately and delete it from your system(s). Internet emails are not necessarily secure. While we take every care, University of Plymouth accepts no responsibility for viruses and it is your responsibility to scan emails and their attachments. University of Plymouth does not accept responsibility for any changes made after it was sent. Nothing in this email or its attachments constitutes an order for goods or services unless accompanied by an official order form.

Zhao900429, 2018-09-06:
> @Zhao900429<https://github.com/Zhao900429> I'm glad that resolved your issue. It seems like there must be a version capability issue with AQWA and BEMIO. Would you mind sharing your files so that we can reproduce the error and resolve the issue in the current version of WEC-Sim? Thank you!
>
>
> [http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

Hi,
I am pleased to share the AQWA output, I attached them in this email. You can try to read it by the WEC-sim 3.0 function.

All the best,
Chenyu

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

kmruehl, 2018-09-06:
@Zhao900429 Unfortunately it doesn't look like the attachment went through via email. Can you please attach it directly to the issue page on GitHub (https://github.com/WEC-Sim/WEC-Sim/issues/262)? Thank you.

Zhao900429, 2018-09-10:
发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

Zhao900429, 2018-09-12:
hi,

Have you revieve the attachment in the last mail?  And do you think is it possible that multi buoys can connnet with  a single PTO-sim system in WEC-sim 3.0?

All hte best,

Chenyu

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

yuyihsiang, 2018-09-12:
Hi Chenyu,

Please directly attach the file to this issue page as suggested by Kelley if she hasn't received the file. 

Regarding multi buoys connecting to  a single PTO (PTO-Sim?) system, from modeling prospective, yes you can. I have done something similar for two WECs for another project. However, I did that using WEC-Sim and SimScape Fluids instead of PTO-Sim. The required time-step size can be small if you use PTO-Sim, which may significantly slow down your simulation.

Thanks, 

YI-Hsiang

kmruehl, 2018-09-18:
@Zhao900429 can you email me the AQWA files at kelley.ruehl@sandia.gov? thank you

kmruehl, 2019-02-01:
@Zhao900429 responding to your email about the following issue:

> In the WECSIM 3.1, I found there may be something wrong with the Read AQQA but I am not sure about that. Would you please give me a favour? The version of the Workbench is 19.0 and the MATLAB is R2016a. The error information could be seen in the following:

```
Undefined function 'startsWith' for input arguments of type 'cell'.

Error in Read_AQWA (line 45)
    TF  = startsWith(raw1(n),'*');

Error in bemio (line 11)
hydro = Read_AQWA(hydro,'D:\paper\two body WEC\wec-sim\twobuoy.AH1','D:\paper\two body WEC\wec-sim\twobuoy.LIS');
```
It appears that this is a path issue. The path you selected has spaces. and that often causes issues. You should be able to change `\two body WEC\` to `\two_body_WEC\` to remove the spaces and run BEMIO without issue. 

Alternatively, I was able to run BEMIO for the AQWA files you sent by replacing the following line in your `bemio.m` file: `hydro = Read_AQWA(hydro,'D:\paper\two body WEC\wec-sim\twobuoy.AH1','D:\paper\two body WEC\wec-sim\twobuoy.LIS');`

with: `hydro = Read_AQWA(hydro,'twobuoy.AH1','twobuoy.LIS');`
This requires the `bemio.m` file be placed in the same directory as your AQWA LIS and AH1 files, as shown below.

[image]

closed 2019-02-01 by kmruehl