WEC-Sim/WEC-Sim #263: Tool for Pressure Differential Devices - "Panel Selection Tool"

HathiHathi, 2018-08-28:
Hi,

I have created an extra functionality to BEMIO that allows, for example, to calculate hydrodynamics of a pressure differential device, such as [the AWS](https://www.oceannews.com/news/energy/aws-and-trident-energy-collaborate-to-control-pressure-in-new-innovative-wave-energy-device). Now I would like to know if it would be desirable/interesting to add this function to the BEMIO code. My worry is that my code, for now, only works with AQWA. If someone thinks this could be a nice addition, I would be happy to share my code to the BEMIO project. Also my first time working with github, so if anyone has some good advice, let me know!

More detail: The tool reads potential output per panel from the AQWA file. And allows to select certain panels to calculate the hydrodynamics over. This way you can exclude for example the bottom panels that are stationary, when the rest of the heaving body moves up and down. The tool calculates all the hydrodynamic matrices and then produces the same type of normalised output as BEMIO does, ready for creating the .h5 file.

Gr,
Joris

HathiHathi, 2018-09-05:
Hi Yi-Hsiang,

Thank you for your reaction, I will consider doing a pull request.

About the changes in BEMIO: Additional to loading the standard AQWA output into BEMIO, I now also loaded the Potentials (~Froude Krylov~, radiation and diffraction potentials) calculated by AQWA into BEMIO. (Froude Krylov potentials are calculated by the code itself for every panel). With these potentials the code calculates the same hydrodynamic coefficients as AQWA but now we can adjust what areas of the object to take into account. For example, when simulating a fully submerged vertical smaller cylinder that slides within another fully submerged vertical larger cylinder: for these two bodies two separate AQWA calculations can be performed. Where the objects overlap with each other, there we want to extract the panels from the hydrodynamic calculations. Because at this overlapping sides, there is no contact with the water, and we want to model the PTO forces there.  
So the code will select what panels to take into account (for example,a selection of all panels except for the panels with a normal z-direction equal to -1). With this selection of panels it calculates the added mass, radiation damping and force RAO's of the object just like AQWA does (if selecting all panels: results are the same to those from AWQA). From there excising code is used: The two bodies are combined into one h5 file and ready for WECSim simulation. 
Excluding the bottom panels from a body can make a large difference to the resulting hydro coefficients, for certain frequencies. That's why this is interesting for wave energy: it has the resonance advantages of a floating buoy and the sheltered advantages of a submerged buoy. 

Is this explanation making it more clear or more fuzzy? Let me know.

Nevertheless, thanks for reading,

Joris

yuyihsiang, 2018-09-12:
Hi Joris,

I think we have a little better understanding on what you did. So I would say do a pull request and we can take a look and probably will have to work with you a little more if we decide to merge it.

Thanks,

Yi-Hsiang

commit 08cad585 (2018-09-05): updated wecsim lib to 2015b, resolves #263
closed 2018-10-11 by kmruehl