WEC-Sim/WEC-Sim #264: WEC-Sim block in Simulink problem

balboaf, 2018-08-30:
Hi, 

I just intalled Matlab v.9.0; Simulink v8.7; Simscape v.4.0 and Simscape Multibody v.4.8. 

Run the path without problems,
[image]

But, when i go to Simulink interface does not appear the WEC-Sim block 
[image]

I tried doing as directed in the "Issues #227", but fails to fix it.

Thanks you,
Best regards!

yuyihsiang, 2018-09-04:
Supposedly, WEC-Sim simulink Lib should work with MATLAB 2015b. We'll check it again asap. On the other hand, we recommend running WEC-Sim in MATLAB 2016b or newer. 

Thanks,

Yi-Hsiang

kmruehl, 2018-09-04:
@balboaf I see that the PTO-Sim library appears, so that makes it especially bizarre that the WEC-Sim library did not. Did you try refreshing the Simulink library (F5) and/or restarting MATLAB? Also, can you open the WEC-Sim RM3 and OSWEC [tutorials](https://github.com/WEC-Sim/WEC-Sim/tree/master/tutorials) *.slx files?

@yuyihsiang FYI, I believe v9.0 is MATLAB 2016a.

yuyihsiang, 2018-09-04:
Based on MATLAB website:
https://www.mathworks.com/products/compiler/matlab-runtime.html
v9.0 is MATLAB 2015b and v9.0.1 is MATLAB 2016a.

kmruehl, 2018-09-05:
@balboaf the issue was that the WEC-Sim library was saved in a newer version of MATLAB (2016b). Unfortunately the WEC-Sim library was not backward compatible with the version of MATLAB you have installed, so I just saved the WEC-Sim library as an older version of MATLAB and it should be compatible with MATLAB 2015b and newer.

commit 91f0a4c4 (2018-09-05): updated wecsim lib to 2015b, resolves #264
closed 2018-09-05 by kmruehl