WEC-Sim/WEC-Sim #266: Fluid Memory - R2018a

BenHowey, 2018-09-04:
I have just updated to MATLAB 2018a and updated to WECsimv2.  I am trying to run the RM3 example and run into the following error:
`Error using error_for_persistent_vars_in_ct_mlfcn (line 9)
Fluid Memory uses constructs that are invalid when the block specifies or inherits a continuous sample time. Invalid constructs include the use of
persistent or global variables, calls to exported functions or using coder.extrinstic.

Error in sfprivate (line 15)
       feval(fcnName,varargin{:});

Error in wecSim (line 217)
sim(simu.simMechanicsFile);
Suggested Actions:
    • Set sample rate of this block to inherit the fastest discrete rate in the model. - Fix
    • Open the properties dialog of this block to set the sample time manually. - Open`

It runs OK on R2016b.

Thanks

yuyihsiang, 2018-09-04:
WEC-Sim has been updated to v3 for a period of time. Please use the latest version (v3) and it has been tested under R2018a.

Thanks,

Yi-Hsiang

yuyihsiang, 2018-09-12:
Did that solve your issue?

Thanks

Yi-Hsiang

BenHowey, 2018-09-20:
Thanks Yi-Hsiang,

Yes that did sort out my problem.  I was wondering ifI could trouble you
further with one more question?

I am trying to model a spar buoy at 1:40 model scale.  When I run the
simulation with the moordyn file attached it crashes Matlab.

Is there something wrong with this input file?  I am very stuck!!

Thanks for your time.

Kind Regards

Ben

--

Mooring line data file for MoorDyn in Lines.dll
---------------------- LINE DICTIONARY -----------------------------------------------------
LineType  Diam    MassDenInAir    EA          BA/-zeta     Can     Cat    Cdn     Cdt 
(-)       (m)       (kg/m)        (N)         (Pa-s/-)     (-)     (-)    (-)     (-) 
chain      0.0045   0.0076     583.376E6        -0.8       1.0     0.0    1.6     0.05
Dyneema    0.0025 0.000133     300.376E6        -0.8       1.0     0.0    1.6     0.05
---------------------- NODE PROPERTIES -----------------------------------------------------
Node    Type      X           Y         Z        M        V        FX       FY       FZ     CdA     CA
(-)     (-)      (m)         (m)       (m)      (kg)     (m^3)    (kN)     (kN)     (kN)   (m^2)   (-)
1       Vessel   -0.0482   -0.1420     0.00       0         0         0        0       0       0     0
2       Fix      -1.6875   -4.9714    -2.00       0         0         0        0       0       0     0
3       Vessel    0.1471    0.0293     0.00       0         0         0        0       0       0     0
4       Fix       5.1491    1.0242    -2.00       0         0         0        0       0       0     0
5       Vessel   -0.0989    0.1128     0.00       0         0         0        0       0       0     0
6       Fix      -3.4616    3.9471    -2.00       0         0         0        0       0       0     0
7       Connect  -0.6395   -2.1050   -0.500     0.0255   0.00017      0        0       0    12.566   1
8       Connect   2.1803    0.4338   -0.500     0.0255   0.00017      0        0       0    12.566   1
9       Connect  -1.4658    1.6712   -0.500     0.0255   0.00017      0        0       0    12.566   1
10      Connect  -0.3215   -0.9470    -1.00      1.807   0.00032      0        0       0    12.566   1
11      Connect   0.9807    0.1950    -1.00      1.807   20.6514      0        0       0    12.566   1
12      Connect  -0.6593    0.7518    -1.00      1.807   20.6514      0        0       0    12.566   1
13      Connect  -1.5182   -4.4722   -1.750   0.000008  1.5625e-09    0        0       0    12.566   1
14      Connect   4.6322    0.9213   -1.750   0.000008  1.5625e-09    0        0       0    12.566   1
15      Connect  -3.1140    3.5510   -1.750   0.000008  1.5625e-09    0        0       0    12.566   1
---------------------- LINE PROPERTIES -----------------------------------------------------
Line     LineType  UnstrLen  NumSegs   NodeAnch  NodeFair  Flags/Outputs
(-)      (-)       (m)         (-)       (-)       (-)       (-)
1        chain      1.1550         8        2        13        tp
2        Dyneema    2.5000         8       13         7        tp
3        Dyneema    1.2230         5        7        10        tp
4        Dyneema       1.0         5       10         1        tp
5        chain      1.1550         8        4        14        tp
6        Dyneema    2.5000         8       14         8        tp
7        Dyneema    1.2230         5        8        11        tp
8        Dyneema       1.0         5       11         3        tp
9        chain      1.1550         8        6        15        tp
10       Dyneema    2.5000         8       15         9        tp
11       Dyneema    1.2230         5        9        12        tp
12       Dyneema       1.0         5       12         5        tp
---------------------- SOLVER OPTIONS-----------------------------------------
0.0001   dtM          - time step to use in mooring integration
0        WaveKin      - wave kinematics flag (0=neglect, the only option currently supported)
3.0e6    kBot         - bottom stiffness
3.0e5    cBot         - bottom damping
2       WtrDpth      - water depth
5.0      CdScaleIC    - factor by which to scale drag coefficients during dynamic relaxation IC gen
0.001    threshIC     - threshold for IC con
0        WriteUnits   - option to skip units line in output files if zero
-------------------------- OUTPUTS --------------------------------
FairTen4 FairTen5 FairTen6
--------------------- need this line ------------------

yuyihsiang, 2018-09-25:
What is the error message you got when MATLAB crashed? Or it just crashed? You may want to reboot your machine and MATLAB. Sometime that helps particularly if you are using a Mac.

I also noticed that your line.txt file has a mooring properties for a full scale case. You may want to modify your line properties. Let me know if that works or not.

Yi-Hsiang

BenHowey, 2018-09-25:
> Thanks Yi-Hsiang,
>
> Which is the properties at full scale?  I through I had scaled everything
> down to 1:40 scale in the lines.txt
>
> Thanks
>
> Ben

Hi Yi-Hsiang,

I'm sorry to bombard you with questions but I have one more question
regrading BEMIO to wecSim.

I am trying to model the device describe above in a channel to represent
tank side walls during experimentation to look at the effects of the walls
on the motion response of the device.

I am using Nemoh to calculate the hydrodynamic coefficients for the mesh
attached (This is 1:40th scale).  I try and run Bemio to create the .hdf5
file and I get the following error:

Index in position 1 exceeds array bounds (must not exceed 6).

Error in Write_H5 (line 82)
        H5_Create_Write_Att(filename,['/body' num2str(i)

'/hydro_coeffs/radiation_damping/state_space/it'],permute(hydro.ss_O((n+1):(n+m),:),[2
1]),'Order of state space
        realization','');

As a side note:  I can plot out the results that look correct using
Plot_BEMIO(hydro)

Is the next step to model the walls as hydrodynamic bodies connected to the
seabed in Simulink and then running wecSIM?

Any help would be amazing!

Thanks

Ben

yuyihsiang, 2018-09-25:
I don't think you have scaled the EA value.

It seems to me that your error message is something to do with the state-space coefficient calculation. Does it go away when you comment out the "hydro = Radiation_IRF_SS(hydro,[],[]);"?

Regarding your last question, I am not sure what exactly you are trying to simulate. It might be helpful if you n provide a  figure to explain that.

Best,

Yi-Hsiang

kmruehl, 2018-10-11:
@BenHowey does this resolve your issue?

closed 2018-10-16 by kmruehl

BenHowey, 2018-10-17:
Hi,

I solved this by manually adding zeros to the SS_O matrix to increase the size to match the other matrices (i.e 12x12 - for two bodies).  Then the write function worked.

Thanks

Ben

almuzakki, 2018-11-22:
Dear @yuyihsiang and @kmruehl ,

I am using MATLAB R2018a and just downloaded (probably) the latest version of wecsim from github.
last time (more than 6mo ago) i tried to run my simulation (using wecsim v2 and matlab r2017a) and it worked fine.

But now, i got the same error as @BenHowey :
"Error using error_for_persistent_vars_in_ct_mlfcn (line 9)
Fluid Memory uses constructs that are invalid when the block specifies or
inherits a continuous sample time. Invalid constructs include the use of
persistent or global variables, calls to exported functions or using
coder.extrinstic.
Error in sfprivate (line 15)
       feval(fcnName,varargin{:});
Error in wecSim (line 257)
sim(simu.simMechanicsFile);
Suggested Actions:
    • Set sample rate of this block to inherit the fastest discrete rate in the
    model. - Fix
    • Open the properties dialog of this block to set the sample time manually.
    - Open"

-----------------------------------------------

I even tried to reboot my PC and MATLAB, it did not work.

Can you help me?

Best,
Zaki