WEC-Sim/WEC-Sim #267: geometry definition when using nemoh as hydroData input

waveswaves, 2018-09-07:
I have run a flat plate geometry in nemoh (6DOF) and calculated RAOs, most interested in ptich. I've then run the geometry in WEC-Sim with regular waves at a specified angular frequency with a 6DOF constraint located at the center of gravity of the plate [0 0 0]. The same Inertias and mass was used. The pitch response using added mass and damping from nemoh doesnt match the steady state response in WEC-Sim. I am not sure why this woud be. Are there certain ways to define the geomtry if using nemoh for WEC-Sim? The reason I ask is that when I look at the Mechanics Explorer animation the centre of mass is not where I specified it in nemoh or WEC-Sim, which was [0 0 0] for both programs. I am attaching a picture for clarification. [0 0 0] has been defined in nemoh and in the .stl file at the center of the geomerty. 
Are there any other things I should be aware of when using nemoh for WEC-Sim? 
[image]

kmruehl, 2018-09-10:
@waveswaves would you be willing to ZIP your WEC-Sim model and share it with us so we can help resolve this issue? Thanks.

waveswaves, 2018-09-11:
Thank you for the help, however I have managed to find my own problem and it was not related to wecSim.

kmruehl, 2018-09-11:
@waveswaves glad to hear that.

closed 2018-09-11 by kmruehl