WEC-Sim/WEC-Sim #268: Query regarding simulations to carry out the body motion about any point of rotation than wrt centre of gravity.

POGULURI, 2018-09-09:
Hi, I am Sunny doing post doctoral in Jeju national University. I need small clarifications in regards to the body motion analysis using WEC-sim. I have a plan to use the WEC-sim if this can do what we want. The briefly explained it below hope some in responds:

We are working on wave energy converter on Salter's duck. Where the duck rotations about a it's centre of rotation than about centre of gravity. These options are available to simulate in WAMIT by using constraint body. Where in WEC-Sim webinar presentation I can see mentioning about body motions about it's centre of gravity. But some of options available in WEC-Sim has constrained motion too. 

To finally say in one sentence can we simulate time domain analysis of a body with any centre of rotation than about centre of gravity using WEC-sim.

Please do clarify it would be great help.

Thanks,
Sunny Kumar.

kmruehl, 2018-09-10:
@POGULURI I believe your question is whether WEC-Sim can be used to simulate a WEC that rotates about a point other than its center of gravity. If that's the question, then the answer is yes. An example of this is provided in the WEC-Sim [OSWEC tutorial](http://wec-sim.github.io/WEC-Sim/tutorials.html#oscillating-surge-wec-oswec). The flap rotates around a hinge that is not located it's at center of gravity - this is done by using a [rotational constraint/pto](http://wec-sim.github.io/WEC-Sim/code_structure.html#constraints). However, as started in the webinar, all body motions are output by WEC-Sim relative to a body fixed coordinate system located body's center of gravity. I hope this helps clarify your question/

POGULURI, 2018-09-11:
Thank you so much for response.

kmruehl, 2018-09-11:
@POGULURI you're welcome. Hope it helps.

closed 2018-09-11 by kmruehl