WEC-Sim/WEC-Sim #269: Run WEC-Sim with MoorDyn in Linux

nbcm, 2018-09-10:
I have tried to run examples RM3_MoorDyn but it did not works well in my machine (CentOS 7). I use the rm3.h5 file which built from RM3 case in tutorials folder. 

In the first try, I got following error information:
 Running MoorDyn (v1.01.00C, 2016-04-20)
   Copyright (c) Matt Hall, licensed under GPL v3.
Error: unable to open lines.txt file

Then I back to check the source code of MoorDyn and it seems the input file in the latest version should be in folder named 'Mooring' instead of 'mooring' (MoorDyn.cpp, line 287).

Then I change the folder's name to 'Mooring' in 'RM3_MoorDyn' and run again. The new issue comes and the error information shows:

Simulating the WEC device defined in the SimMechanics model RM3MoorDyn.slx...   

 Running MoorDyn (v1.01.00C, 2016-04-20)
   Copyright (c) Matt Hall, licensed under GPL v3.
   Creating mooring system.  3 fairleads, 3 anchors, 3 connections.
   Warning from Catenary: Unstretched line length too large.
   Warning from Catenary: Unstretched line length too large.
   Warning from Catenary: Unstretched line length too large.
   Finalizing ICs using dynamic relaxation (5X normal drag)
    t = 1 s, tension at first fairlead is 77716.5 N    
    t = 2 s, tension at first fairlead is 1.01718e+06 N    
    t = 3 s, tension at first fairlead is 199041 N    
    t = 4 s, tension at first fairlead is 186692 N    
    t = 5 s, tension at first fairlead is 175286 N    
    t = 6 s, tension at first fairlead is 175517 N    
    t = 7 s, tension at first fairlead is 174200 N    
    t = 8 s, tension at first fairlead is 174005 N    
    t = 9 s, tension at first fairlead is 174790 N    
    t = 10 s, tension at first fairlead is 174934 N    
    t = 11 s, tension at first fairlead is 174871 N    
   Fairlead tensions converged to 0.1% after 11 seconds.        

Error using wecSim (line 257)
Error in 'RM3MoorDyn/MoorDyn': Initialization commands cannot be evaluated.
Caused by:
    Error using textread (line 159)
    File not found.

Is there any path setting incorrect in WEC-Sim? Anyone has the same issue and can help?

yuyihsiang, 2018-09-12:
Hi,

The file names could be a problem for Linux. It seems that the code can not file a file. I think I found your issue. However, I don't have a Linux machine to test. For now, try to change line 88 in the mooringClass.m file from "mooring" to "Mooring" (see the link below). 

[image]

Let us know if that fix your problem. If it does work, I'll modify our code.

Thanks,

Yi-Hsiang

nbcm, 2018-09-14:
Now the MoorDyn worked but still have some issues in post-processing and saving.

 Running MoorDyn (v1.01.00C, 2016-04-20)
   Copyright (c) Matt Hall, licensed under GPL v3.
   Creating mooring system.  3 fairleads, 3 anchors, 3 connections.
   Warning from Catenary: Unstretched line length too large.
   Warning from Catenary: Unstretched line length too large.
   Warning from Catenary: Unstretched line length too large.
   Finalizing ICs using dynamic relaxation (5X normal drag)
    t = 1 s, tension at first fairlead is 77716.5 N    
    t = 2 s, tension at first fairlead is 1.01718e+06 N    
    t = 3 s, tension at first fairlead is 199041 N    
    t = 4 s, tension at first fairlead is 186692 N    
    t = 5 s, tension at first fairlead is 175286 N    
    t = 6 s, tension at first fairlead is 175517 N    
    t = 7 s, tension at first fairlead is 174200 N    
    t = 8 s, tension at first fairlead is 174005 N    
    t = 9 s, tension at first fairlead is 174790 N    
    t = 10 s, tension at first fairlead is 174934 N    
    t = 11 s, tension at first fairlead is 174871 N    
   Fairlead tensions converged to 0.1% after 11 seconds.        

   MoorDyn closed.
Elapsed time is 113.668019 seconds.

Post-processing and saving...   
Error using fgets
Invalid file identifier. Use fopen to generate a valid file identifier.

Error in fgetl (line 32)
[tline,lt] = fgets(fid);

Error in responseClass/loadMoorDyn (line 96)
            header = strsplit(fgetl(fid));

Error in postProcess (line 73)
        output.loadMoorDyn(mooring(iMoor).moorDynLines);

Error in wecSim (line 265)
postProcess

nbcm, 2018-09-14:
It seems we need also change the 'responseClass.m' 
line 94
line 107

Then it works well on my machine (CentOS7)

 Running MoorDyn (v1.01.00C, 2016-04-20)
   Copyright (c) Matt Hall, licensed under GPL v3.
   Creating mooring system.  3 fairleads, 3 anchors, 3 connections.
   Warning from Catenary: Unstretched line length too large.
   Warning from Catenary: Unstretched line length too large.
   Warning from Catenary: Unstretched line length too large.
   Finalizing ICs using dynamic relaxation (5X normal drag)
    t = 1 s, tension at first fairlead is 77716.5 N    
    t = 2 s, tension at first fairlead is 1.01718e+06 N    
    t = 3 s, tension at first fairlead is 199041 N    
    t = 4 s, tension at first fairlead is 186692 N    
    t = 5 s, tension at first fairlead is 175286 N    
    t = 6 s, tension at first fairlead is 175517 N    
    t = 7 s, tension at first fairlead is 174200 N    
    t = 8 s, tension at first fairlead is 174005 N    
    t = 9 s, tension at first fairlead is 174790 N    
    t = 10 s, tension at first fairlead is 174934 N    
    t = 11 s, tension at first fairlead is 174871 N    
   Fairlead tensions converged to 0.1% after 11 seconds.        

   MoorDyn closed.
Elapsed time is 114.758358 seconds.

Post-processing and saving...   
Warning: MATLAB has disabled some advanced graphics rendering features by switching to software OpenGL. For more information, click
here. 
Elapsed time is 9.310052 seconds.

yuyihsiang, 2018-09-25:
Great, thanks for the info. we will fix that.

Thanks,

Yi-Hsiang

yuyihsiang, 2018-10-04:
Updated

closed 2018-10-04 by yuyihsiang

efantnu, 2018-10-17:
Hi,
thanks for this post.

I retrieved the repository today and tried to run a simulation in my Linux machine (it works in a Windows machine).

I changed the name of lines.txt folder to "Mooring", as indicated. But I still had to change line 88 in mooringClass.m to:
`obj.moorDynInputRaw = textread('./Mooring/lines.txt', '%s', 'delimiter', '\n');`

Everything went smoothly after that.

It that supposed to be so?