WEC-Sim/WEC-Sim #273: Negative added mass results in negative mass and simualtion failure

crobarcro, 2018-10-19:
I have a submerged body for which the Nemoh simulations result in the added mass being negative in the z direction at some frequencies. See the plot below which was produced using the command:

`plot (1./(obj.hydroData.simulation_parameters.w./tau()), squeeze(am(ii,jjj,:)))`

when stopped in the `constAddedMassAndDamping` method of `bodyClass` where the `obj.hydroForce.fAddedMass` field is being populated.

[image]

This is for a roughly cylindrical non-surface piercing fully submerged buoy about 2m below the mean water level.

When I try to simulate this system in a regular wave of Hs 2.5 and period 8s I have the problem that the body mass becomes negative as line 193 of `bodyClass` in the `adjustMassMatrix` method contains the following:

`tmp.adjmass = sum(tmp.fadm(1:3))*adjMassWeightFun;`

the problem is that the sum of the added masses is negative and greater in magnitude than the actual mass of the buoy. So my question is, do you think I have made a mistake in my data generation with Nemoh (I don't think I have my input parameters wrong, and it's a pretty simple body), or is this a WEC-Sim issue? If it is a WEC-Sim issue can you suggest a workaround?

crobarcro, 2018-10-28:
I realised this is only an issue for single frequency simulations using constant added mass and damping (the 'regular' wave option).

nathanmtom, 2018-11-02:
@crobarcro Thank you for the question regarding negative added mass. We agree that when running the 'regular' wave option that when summing the added mass and the mass the magnitude might drop below 0. We also suggest plotting the wave damping values and ensure that these values do not drop below 0 which can also lead to WEC-Sim becoming unstable. Based on your second comment, did the simulation run when you use the 'regularCIC' wave option?

It is difficult to say if you have an error in the NEMOH input files; however, assuming your input files are correct it is possible that NEMOH, or other BEM solver (i.e. WAMIT), may produce negative added mass values. This has been found to occur more frequently for non-surface piercing fully submerged bodies that are close to the free-surface relative to the bodies characteristic length. Thus, if the radius of your buoy is on the same order of magnitude or greater than the submergence depth NEMOH may report negative added mass values. For more information see, McIver, P. and Evans, D.V., 1984, "The occurance of negative added mass in free-surface problems involving submerged oscillating bodies," Journal of Engineering Mathematics, 18, pp 7-22.

In addition, based on the shape of your plot, the abrupt change between the positive and negative added mass values observed at the 8.4 s wave period does resemble the characteristics of an irregular wave frequency. This phenomena is related to how the BEM solution is calculated at certain wave frequencies. There are techniques to change the calculation method at these frequencies; however, at the moment NEMOH doesn't have this capability. WAMIT does have this option and I suggest reviewing the latest WAMIT manual which provides additional background information on irregular wave frequencies. 

I hope these answers are sufficient to address your own model issues. If you have additional questions or want clarification on the above comments, please let me know.

Cheers,
Nathan Tom
National Renewable Energy Laboratory

yuyihsiang, 2018-11-09:
Hopefully, we have answered your question. Let us know if you still have issue on this.

Thanks,

Yi-Hsiang

crobarcro, 2018-11-09:
Yes, it's answered thanks!

closed 2018-11-09 by crobarcro