WEC-Sim/WEC-Sim #276: possible to define body to body mechanical coupling

Tudore, 2018-11-19:
Hi, Thank you for your great effort to WEC-SIM. It is a very powerful tool not only for WEC but also for Multibody interactions in other applications.

I am now working for my PhD thesis. I would like to make it read WAMIT -.8,-.9,-.7. I think it can be used for calculating drift force in the time domain. 

The other question is, can we define, a multibody system with mechanical coupling between them. For example, I have an array of three bodies connected through springs. And the last body is connected to the seabed. Is this possible to be defined in WEC-SIM now？

Tudore, 2018-11-19:
Another setup is I have several bodies, all these bodies are connected to one body, and this end body is connected to the base.

yuyihsiang, 2018-11-30:
Hi Tudore,

Sorry for taking so long to replay. It was Thanksgiving Holiday in the US last week and most of us were out.

I think your question includes two parts:

1.  The drift force calculation is one of the request feature listed in the Projects.  I believe reading those WAMIT files is one way to calculate the drift force, and we will be very appreciated if you decide to incorporate it into BEMIO. Note that if I understand correctly, WAMIT added additional drift force calculation capability and changed the output file name when upgrading from v6.X to V7.X. 

2. To your second question: Yes, I believe what you have said is very possible. You just have to have all the bodies modeled in WAMIT. In this case, you should have 3 bodies in WAMIT. In WEC-Sim, I think you can use 3 body-blocks and connect them with spring, e.g., translational PTOs with a given pto.k values (assuming the springs are pre-tensioned and the tension is always larger than 0). 

Let me know if you have any question. You can also upload your case if you have issue setting it up. 
  
Best,

Yi-Hsiang

Tudore, 2018-12-02:
@yuyihsiang Thank you very much for your reply. I also noted that it was Thanksgiving season in the U.S.A.
 (1) About WAMIT drift force, I am modifying BEMIO now. You are right, there are some changes in WAMIT output files. Now I am using WAMIT-v7.2. I will keep on updating the progress if any in the next several months.
 (2) For my second question, I have already built the three-body model in WAMIT as well as in WEC-SIM. I tried to use PTOs to connect my body-2 and body-3 to body-1. However, it seems if I apply PTO, it will restrain all the other D.O.F of motion except for the D.O.F in heave (Z). Let's say, I have a PTO between body-2 and body-1, then, the motions in (X, Y, and Rx, Ry, Rz ) are restrained. That is to say, body-2 is only allowed to move at z direction relative to body-1.  The expected case is the body in two directions (x,y) with restoring stiffness, but the remaining directions no restraints. 

BTW, the pretension is an interesting function as well. Is there any means to apply pretensions for the springs?

Thanks again for your so nice help!

yuyihsiang, 2019-01-08:
Sorry for the late reply. I was out for Christmas and New year vacation as well. 

Regarding your second question, it will be easier for us to understand, if you can upload your model and a figure that describe the problem and the required DOFs in a greater details.

Thanks,

Yi-Hsiang

Tudore, 2019-01-10:
Hi, Yi-Hsiang, 
It is totally understood. Happy new year! This is the season for holiday.
I will prepare more detailed information to you later.
Thanks!
Zhang Chi

Tudore, 2019-02-13:
Hi, Again,

I am still testing my model. May I check with you why WAMIT model should have a reference point at its COG while the others are referring to the mean free surface? In WAMIT, the default reference point is also on free surface.

Thanks!
Zhang Chi

yuyihsiang, 2019-03-04:
We need to have an origin at CG because of the equation of motion for the WEC in WEC-Sim is solved at CG.

Yi-Hsiang

Tudore, 2019-03-05:
Can this be solved by transfering WAMIT results from Reference point at
Free surface to Reference point at CG in Bemio?

On the other word, can we transfer WEC-SIM results from COG to free
surface?

The middle of Aw seems to be a popular reference for hydrodynamics.

Regards,
Zhang Chi

kmruehl, 2019-03-05:
@Tudore BEMIO is a input/output parser that reads the WAMIT output file (`*.out`) and parses it into an HDF5 file (`*.h5`) that is later imported by WEC-Sim. BEMIO does not transfer WAMIT results (or any other BEM code's results) from one reference point to another (i.e. from the free surface to the cg). The recommended workflow is to solve for the WEC hydrodynamic coefficients in WAMIT with the origin at the CG, and then run BEMIO to generate the `*.h5` file of WEC hydrodynamic coefficients at the CG. As @yuyihsiang said, this is necessary because WEC-Sim solves the equation of motion for each body about its cg (not about the free surface).

closed 2019-03-05 by kmruehl