WEC-Sim/WEC-Sim #277: Store Simulink data in the "parent" workspace of MATLAB

efantnu, 2018-11-22:
WEC-Sim is being used for simulations in the [IMAGINE project](https://ec.europa.eu/inea/en/horizon-2020/projects/h2020-energy/tidal-wave-hydropower/imagine).

We are using fminsearch from the Optimization toolbox of Matlab to find optimal parameter sets for the EMG controller. One challenge with that is the Simulink part of WEC-Sim saves/reads all variables in/from the [base workspace] (https://se.mathworks.com/help/matlab/matlab_prog/base-and-function-workspaces.html) of MATLAB and those are not readable/writable when wecSim.m is called inside a function. 

An external workaround for that results in really poor programming and larger simulation times. So, we did a minor change in line 257 of wecSim.m to fix that:
```matlab
sim(simu.simMechanicsFile, [], simset('SrcWorkspace','parent'));
```

We kindly ask you to include this in the official release, so others can benefit from that too.

PS: This does not affect the normal behavior of WEC-Sim when called from a script or the command line, because then the parent workspace will be the base workspace.

kmruehl, 2018-11-27:
@efantnu this sounds like an excellent suggestion, can you resubmit this as a [pull request](https://github.com/WEC-Sim/WEC-Sim/pulls) so that you can be credited for this code revision? Thank you.

efantnu, 2018-12-07:
@kmruehl, I tried to publish my changes today, but got this message:

> remote: Permission to WEC-Sim/WEC-Sim.git denied to efantnu.
> fatal: unable to access 'https://github.com/WEC-Sim/WEC-Sim/': The requested URL returned error: 403

I also tried to create a new branch "parent-workspace" but that did not work as well.

PS: My knowledge of git and git-hub is very limited, so maybe I am doing something wrong. I tried all that via Atom editor.

kmruehl, 2018-12-13:
@efantnu No worries, you'll have to submit a pull request through the WEC-Sim GitHub repository here: https://github.com/WEC-Sim/WEC-Sim/pulls. How are you accessing WEC-Sim now? Do you have a fork of the repository?

efantnu, 2018-12-17:
Ok, I think I managed to do it now.

I had not forked, and was trying to commit my changes directly to the main repository. Dummies mistake. I wold say that was a good git / GitHub crash course :-D

kmruehl, 2018-12-20:
No worries! Thank you for submitting the pull request, I'll take a look at it after the new year.

closed 2018-12-20 by kmruehl