WEC-Sim/WEC-Sim #278: Updates of PM sea state generation in version 3.0

annetbr, 2018-11-28:
Hi, 

I have done some convergence studies for power capture with increasing simulation lengths. Simulations were done with both version 2.1 and 3.0 of WEC-Sim. With version 2.1 of the software, the power seemed to be overpredicted for short simulation run times for all waves simulated. However, with version 3.0 the power was overpredicted in some cases and underpredicted in other cases, which I think is to be expected. Could this phenomenon be due to an update in the software? Were any updates done with regards to how the sea states are generated in the simulation?

Please see the figures with some examples. In the figures the x-axis (Runtime) gives the total run time of each simulation (not time steps of a single simulation). The numbers on the y-axis have been removed due to confidentiality. The simulations were run with a Pierson-Moskowitz wave spectrum with a range of different Hs, Te and wave seeds.  
[image]
[image]

Let me know if you need any additional information. 

Thank you in advance!

Best regards, 

Annette Brask

yuyihsiang, 2018-11-30:
We have changed the method of frequency discretization for irregular waves in V3.0. The default is equal energy in V3.0, but the default is equal frequency spacing in v2.1. You way want to check "waves.freqDisc" variable as well as "waves.numFreq" in both versions. If they are different, that means different irregular waves even if the same random seed is used.

Yi-Hsiang

annetbr, 2018-12-03:
Okay, thanks! Could this explain why the power seems to be over predicted most of the time for short run times with version v2.1 of the software? 

Best regards, 

Annette

yuyihsiang, 2018-12-07:
I believe it is just because the random seed that has been used, which happened to have a smaller waves in the short run with version 2.1. 

Yi-Hsiang

annetbr, 2018-12-13:
Okay, that makes sense. 

Thank you for your reply! 

Annette

closed 2019-01-08 by kmruehl